hathach/tinyusb #2828: Don't call pbuf_free when ethernet_input is Ok

Slion, 2024-10-04:
### Operating System

Others

### Board

Raspberry Pi Pico WH

### Firmware

̀ net_lwip_webserver`

### What happened ?

If you call `pbuf_free` on the frame when `ethernet_input` returns `ERR_OK` you will get either `pbuf_free: p->ref > 0` or worst steal the buffer ownership from the HTTP server.

### How to reproduce ?

Try run `net_lwip_webserver` on Pico.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Sorry won't bother, description is good enough.

### Screenshots

lwIP ticket: https://savannah.nongnu.org/patch/index.php?10121
See example of how it should be done: https://github.com/lwip-tcpip/lwip/blob/master/contrib/examples/ethernetif/ethernetif.c
Here is the faulty code: 
https://github.com/hathach/tinyusb/blob/6bfd02413785b8cdb3b4bb7e2f3851be0a50156e/examples/device/net_lwip_webserver/src/main.c#L198

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-10-04:
Hi, only few of mainteners are good at lwip, do you any suggestion how to fix it ?

Slion, 2024-10-04:
> Hi, only few of mainteners are good at lwip, do you any suggestion how to fix it ?

```C
if (ethernet_input(received_frame, &netif_data)!=ERR_OK) {
    pbuf_free(received_frame);
}
```

HiFiPhile, 2024-10-05:
Thank you , PR is on: #2829