hathach/tinyusb #2830: lnet_lwip_webserver lwIP configuration inconsistency

Slion, 2024-10-05:
### Operating System

Others

### Board

Raspberry Pi Pico WH

### Firmware

`lnet_lwip_webserver`

### What happened ?

OOM errors in lwIP logs.

Enable logs like that:
```
#define LWIP_DEBUG
#define HTTPD_DEBUG LWIP_DBG_ON
//#define TCP_OUTPUT_DEBUG LWIP_DBG_ON
//#define TCP_INPUT_DEBUG LWIP_DBG_ON
#define MEM_DEBUG LWIP_DBG_ON
```

### How to reproduce ?

Enable logs and maybe use some larger web page.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Not so relevant, that's about lwIP usage.

### Screenshots

There is something funny with this example [lwIP configuration](https://github.com/hathach/tinyusb/blob/master/examples/device/net_lwip_webserver/src/lwipopts.h).

It does not specify `MEM_SIZE` which means the default 1600 bytes is used.
Then it specifies `TCP_MSS` as 1500 bytes which will result in a `PBUF_POOL_BUFSIZE` of about the same I believe.
So there is barely enough room for a single pbuf in the lwIP heap.
Also the overriding of `PBUF_POOL_SIZE` from 4 to 16 is questionable I think.

When I built on this example to serve slightly larger web pages needing around 32 GET requests to load it worked mostly.
But when I enabled the lwIP logs it became clear something was not right as there was a lot of OOM warnings.

I would remove `TCP_SND_BUF`, `TCP_WND` and `PBUF_POOL_SIZE` overrides.
Leave `TCP_MSS` as it is maybe and add:

```
// Lower this if you don't want to use so much memory
// You could also comment out TCP_MSS override and use the very conservative default instead
#define MEM_SIZE 1024 * 8 // or 16 maybe?
```

Than again I admit I know very little about the inner working of lwIP myself so we ought to ask the experts really.
However those changes did fix my OOM warnings and improved HTTP performance in my case.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Slion, 2024-10-05:
@goldsimon @yarrick what's your opinion on this? Any good read you recommend about lwIP configuration?

Slion, 2024-10-05:
Related to #2828

yarrick, 2024-10-06:
Try sending a question to lwip-users@nongnu.org and include debug logs and such.

Slion, 2024-10-16:
After doing some tests and monitoring the heap size it's clear pbufs are not allocated from `MEM_SIZE` so you don't really need it and could keep using the default. I'm using 2KB.
That rest of the suggested changes do make sense though.