hathach/tinyusb #2838: The uac2_headset example is broken on RP2040 processors

pschatzmann, 2024-10-13:
### Operating System

Linux

### Board

RP2040

### Firmware

standard example with standard firmware as documented in this project

### What happened ?

The processor is hanging after connecting to the device from Audacity under Linux

After implementing [this pull request](https://github.com/hathach/tinyusb/pull/1802), it's working w/o problems.
How can I help that this correction is moving agead ?
I finally want to have it finally in the Adafuit TinyUSB....

### How to reproduce ?

- Compile and deploy headset example
- select TinyUSB as input device
- start Audacity 
- click recording
- click change sample rate or select TinyUSB as output device

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

I could not get the logging to work 

### Screenshots

n/a

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

FearLabsAudio, 2024-12-22:
This is still broken despite merges ISO API changes #2847 

Pi Pico Device, Pico-SDK 2.1.0, Results are the same on multiple windows hosts as well as linux hosts

Output (CFG_TUSB_DEBUG=1) is as follows:
```
Headset running
Get channel 0 volume range (-50, 0, 1) dB
Get channel 1 volume range (-50, 0, 1) dB
Get channel 2 volume range (-50, 0, 1) dB
Clock get 2 freq ranges
Range 0 (44100, 44100, 0)
Range 1 (48000, 48000, 0)
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Get channel 1 mute 0
Get channel 1 volume 0 dB
Get channel 1 mute 0
Clock set current freq: 48000
WARN: starting new transfer on already active ep 01

*** PANIC ***

ep 01 was already available
```

rahmanshaber, 2025-03-15:
Tried the uac2_headset example in pico2040 with linux, and the device is not showing up and this the output from dmesg. Any help would be appreciated 

```

[  +0.093643] usb 3-2.2: USB disconnect, device number 10
[  +0.434362] usb 3-2.2: new full-speed USB device number 11 using xhci_hcd
[  +0.109792] usb 3-2.2: New USB device found, idVendor=cafe, idProduct=4010, bcdDevice= 1.00
[  +0.000004] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-2.2: Product: TinyUSB headset
[  +0.000002] usb 3-2.2: Manufacturer: TinyUSB
[  +0.000001] usb 3-2.2: SerialNumber: DE60A0F84B775B39
[Mar15 15:14] usb 3-2.2: parse_audio_format_rates_v2v3(): unable to retrieve number of sample rates (clock 4)
[  +5.120017] usb 3-2.2: parse_audio_format_rates_v2v3(): unable to retrieve number of sample rates (clock 4)
[ +10.240000] usb 3-2.2: 2:0: failed to get current value for ch 1 (-110)
[ +10.239984] usb 3-2.2: 2:0: failed to get current value for ch 2 (-110)
[ +10.240022] usb 3-2.2: 2:0: failed to get current value for ch 0 (-110)
[ +10.239989] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119973] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[Mar15 15:15] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.120994] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119020] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119969] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119994] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.121087] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.118937] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119966] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[Mar15 15:16] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[ +20.480023] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.118978] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +0.985385] NVRM: GPU at PCI:0000:01:00: GPU-4a3f58cb-a134-03e2-4555-7dfbdbeee006
[  +0.000005] NVRM: Xid (PCI:0000:01:00): 56, CMDre 00000000 00002888 000100c1 00000007 00000000
[  +0.000665] NVRM: Xid (PCI:0000:01:00): 56, CMDre 00000000 0000288c 000100c1 00000007 00000000
[  +4.133949] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.120052] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.120100] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.119849] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[  +5.120022] usb 3-2.2: 2:0: cannot get min/max values for control 2 (id 2)
[Mar15 15:17] usb 3-2.2: uac_clock_source_is_valid(): cannot get clock validity for id 4
[  +0.000006] usb 3-2.2: clock source 4 is not valid, cannot use
[  +5.119030] usb 3-2.2: 1:1: cannot get freq (v2/v3): err -110
[  +5.119982] usb 3-2.2: 1:1: cannot set freq 48000 (v2/v3): err -110
[  +5.119995] usb 3-2.2: uac_clock_source_is_valid(): cannot get clock validity for id 4
[  +0.000007] usb 3-2.2: clock source 4 is not valid, cannot use
[  +5.121074] usb 3-2.2: 1:1: cannot get freq (v2/v3): err -110
[  +5.118924] usb 3-2.2: 1:1: cannot set freq 48000 (v2/v3): err -110
[  +5.120011] usb 3-2.2: uac_clock_source_is_valid(): cannot get clock validity for id 4
[  +0.000007] usb 3-2.2: clock source 4 is not valid, cannot use
[  +5.119989] usb 3-2.2: 1:1: cannot get freq (v2/v3): err -110
[  +5.119976] usb 3-2.2: 1:1: cannot set freq 48000 (v2/v3): err -110
[Mar15 15:18] usb 3-2.2: uac_clock_source_is_valid(): cannot get clock validity for id 4

```

hathach, 2025-05-21:
should be fixed by https://github.com/hathach/tinyusb/pull/2937

closed 2025-05-21 by hathach