hathach/tinyusb #2842: Example for HID PID device with FFB capabilities

ChrGri, 2024-10-15:
### Related area

Examples

### Hardware specification

ESP32 S3

### Is your feature request related to a problem?

I'd like to built a force feedback (FFB) steering wheel with an ESP32 S3, yet I'm having trouble getting started. 
There are a lot of helpful examples. A HID example with FFB capabilities would be awesome. 

There are also some existing projects for arduino, e.g. https://github.com/hoantv/VNWheel, but I cannot find any project related to ESP32 or tinyusb. 

Any help is very much appreciated. 

### Describe the solution you'd like

A hello world example for a HID device wiith FFB capabilities.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

GalatanuM, 2025-03-10:
Still nothing about this topic?

hathach, 2025-03-10:
I don't have time and experience with ffb to write this example. PR is welcome

closed 2025-03-10 by hathach