hathach/tinyusb #2872: tun_umoun_cb is not called for stm32f4xx

milabs, 2024-11-10:
### Operating System

Linux

### Board

STM32F4XX M

### Firmware

cdc+dwc2

### What happened ?

related to https://github.com/hathach/tinyusb/issues/1476

### How to reproduce ?

cdc + dwc2

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

-

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

milabs, 2024-11-10:
Note, that in my test `USBD Suspend : Remote Wakeup = 0` event does NOT always come on USB cable plug. So, it seems that at least that logic doesn't work well.

HiFiPhile, 2024-11-10:
It's normal that tud_umount_cb() doesn't called on cable unplug, the information is not returned by DCD without Vbus connected.
You need also use tud_suspend_cb() for unplug detection.

milabs, 2024-11-10:
`tud_suspend_cb` is not called every time I plug/unplug either, so it's not reliable as I see ... any ideas? @HiFiPhile

HiFiPhile, 2024-11-11:
> `tud_suspend_cb` is not called every time I plug/unplug either, so it's not reliable as I see ... any ideas? @HiFiPhile

Are you testing with stock examples ? Please post your schematic.

milabs, 2024-11-11:
I'm using the following board:
https://stm32-base.org/boards/STM32F407VGT6-STM32F4XX-M.html

Board's schematic:
https://stm32-base.org/assets/pdf/boards/original-schematic-STM32F407VGT6_STM32F4XX_M.pdf

I don't use stock examples and here us a snippet with USB initialization of my project:

~~~
        long do_init(void) {
                GPIO_InitTypeDef gpio = { 0 };

                __HAL_RCC_GPIOA_CLK_ENABLE();

                //
                // NOTE: We use that code to reset USB on RESET so it'll be
                //       possible to open ttyACM right after flashing
                //

                gpio.Pin = GPIO_PIN_11 | GPIO_PIN_12;
                gpio.Mode = GPIO_MODE_OUTPUT_PP;
                gpio.Pull = GPIO_NOPULL;
                HAL_GPIO_Init(GPIOA, &gpio);

                //
                // 250ms delay is enough to wait after USB reset
                //

                HAL_Delay(250);

                // USB_OTG_FS GPIO Configuration
                // -----------------------------
                //   PA11 => USB_OTG_FS_DM
                //   PA12 => USB_OTG_FS_DP

                gpio.Pin = GPIO_PIN_11 | GPIO_PIN_12;
                gpio.Mode = GPIO_MODE_AF_PP;
                gpio.Pull = GPIO_NOPULL;
                gpio.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
                gpio.Alternate = GPIO_AF10_OTG_FS;
                HAL_GPIO_Init(GPIOA, &gpio);

                // Peripheral clock enable
                __HAL_RCC_USB_OTG_FS_CLK_ENABLE();

                HAL_NVIC_SetPriority(OTG_FS_IRQn, 0, 0);
                HAL_NVIC_EnableIRQ(OTG_FS_IRQn);

                // STM32F407VGT6 doesn't use VBUS sense (B device) explicitly disable it
                USB_OTG_FS->GCCFG |= USB_OTG_GCCFG_NOVBUSSENS;
                USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBUSBSEN;
                USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBUSASEN;

                return HAL_OK;
        }
~~~

HiFiPhile, 2024-11-11:
I'm afraid it's more like a hardware limitation, on STM32F4-DISCO with HS breakout cable I managed to reproduce it (although pretty rare)

Sometimes nothing triggered in DWC2 when cable is unplugged, even not the early suspend flag.

milabs, 2024-11-11:
@HiFiPhile Any API call I can use to force `tinyusb` to switch to disconnected state manually? Let's say I have ping/pong logic implemented on top of my data protocol so I can check if it timed out.

HiFiPhile, 2024-11-11:
> Any API call I can use to force tinyusb to switch to disconnected state manually

There is no dedicated API since the issue is specific to DWC2, but     `tud_deinit(BOARD_TUD_RHPORT);
    tud_init(BOARD_TUD_RHPORT);` should do the job to reset state machine.

milabs, 2024-11-13:
@HiFiPhile Thanks

closed 2024-11-14 by HiFiPhile