hathach/tinyusb #2908: mentor/musb: compilation errors for ARM M0 (cortex-m0plus)

cdesmondee, 2024-12-17:
### Operating System

Linux

### Board

N/A

### Firmware

Any example will not build.

### What happened ?

I am working on an M0 device that is utilizing a Mentor Graphics USB controller. When trying to compile any of the examples the compilation fails highlighting that taking an address of packed struct can result in an unaligned memory access. Specifically for M0 unaligned memory accesses will result in hard faults.

I have been able to get the a subset of the examples working, but I had to hack the source to not utilize structs e.g. tusb_control_request_t and manipulate registers directly.

I have the intention of upstreaming support for this device, so even though these changes are functional, it is not my end goal for my project.

### How to reproduce ?

Not reproducible since the target device is not currently supported.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Not applicable, binary will not compile.

### Screenshots

Compiler Error:
<TINYUSB_INSTALL_PATH>/src/portable/mentor/musb/dcd_musb.c:239:24: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  239 |   uint32_t *p = (void*)&_dcd.setup_packet;
      |                        ^~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2024-12-20:
Can you specify which mcu you are working on ?

cdesmondee, 2024-12-20:
> Can you specify which mcu you are working on ?

I am working on an M0 device that is not public yet. Once it is public I plan on sending a PR for the device so that the device will be supported in mainline.

hathach, 2024-12-25:
can you try to see if this can be fixed by adding TU_ATTR_ALIGNED(4) to the struct around line 76 

```C
typedef struct
{
   TU_ATTR_ALIGNED(4) tusb_control_request_t setup_packet;
  uint16_t     remaining_ctrl; /* The number of bytes remaining in data stage of control transfer. */
  int8_t       status_out;
  pipe_state_t pipe0;
  pipe_state_t pipe[2][TUP_DCD_ENDPOINT_MAX-1];   /* pipe[direction][endpoint number - 1] */
  uint16_t     pipe_buf_is_fifo[2]; /* Bitmap. Each bit means whether 1:TU_FIFO or 0:POD. */
} dcd_data_t;
```

cdesmondee, 2025-01-13:
Thank you for the suggestion. I applied the macro at the location suggested as well as in another place and was able to compile successfully. The issue is resolved. I will close.

closed 2025-01-13 by cdesmondee