hathach/tinyusb #2926: switch between CDC and MSC device on esp32-s3

marchingband, 2025-01-02:
### Related area

device stack?

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

I am working on a device where most of the time, the ESP32 is a USB serial device, but then sometimes, it needs to change into a MSC device for a while, and then back to serial again.
This is to allow users to upload config files, which they will only do sometimes, and which should not be active all the time.
I see the example in the esp-idf for composite cdc and msc device, but if I do not initialize the msc, then then the driver crashes.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420046d8  PS      : 0x00060f30  A0      : 0x82023cb2  A1      : 0x3fc9ff70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060120  A5      : 0x3fca000c  
A6      : 0x0000001f  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3fc9ff60  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00060d23  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420046d5:0x3fc9ff70 0x42023caf:0x3fc9ff90 0x420245b6:0x3fc9ffe0 0x42025e61:0x3fca0000 0x420039c3:0x3fca0030 0x4037d1bd:0x3fca0050

  #0  0x420046d5:0x3fc9ff70 in tud_msc_test_unit_ready_cb at components/espressif_esp_tinyusb_1.4.2_2/tusb_msc_storage.c:525
  #1  0x42023caf:0x3fc9ff90 in proc_builtin_scsi at managed_components/espressif__tinyusb/src/class/msc/msc_device.c:658
  #2  0x420245b6:0x3fc9ffe0 in mscd_xfer_cb at managed_components/espressif__tinyusb/src/class/msc/msc_device.c:471
  #3  0x42025e61:0x3fca0000 in tud_task_ext at managed_components/espressif__tinyusb/src/device/usbd.c:567 (discriminator 2)
  #4  0x420039c3:0x3fca0030 in tud_task at managed_components/espressif__tinyusb/src/device/usbd.h:55 (discriminator 1)
      (inlined by) tusb_device_task at components/espressif_esp_tinyusb_1.4.2_2/tusb_tasks.c:41 (discriminator 1)
  #5  0x4037d1bd:0x3fca0050 in vPortTaskWrapper at /Users/andy/.platformio/packages/framework-espidf@3.50102.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### Describe the solution you'd like

Is there a way to either
a) register as a composite device, but defer initialization of the msc device, and then later deinit it?
b) register as a CDC, then later deinit the CDC device, register as an MSC, init the MSC, and then back again?

Example code for this on ESP32, where menuconfig is so intimately connected to USB driver, would be deeply appreciated.
any tips appreciated really, I have not been able to find any info online, thank you!

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

marchingband, 2025-01-02:
I have worked on the individual initialization sequences such that I can start either the CDC or the MSC, without having to adjust sdkconfig.
However the teardown process is not working.
Digging through the source code I did find
```
// TODO
// bool tusb_teardown(void);
```
Is there a way i can accomplish this teardown without restarting the MCU?

roma-jam, 2025-01-02:
Hi @marchingband, 

The esp-idf is using the [esp-tinyusb](https://github.com/espressif/esp-usb/tree/master/device/esp_tinyusb) wrapper, which provides the configuration and and other things for the classes and simplify the final usage.
In the example for the composite device both classes must be initialized. 

For the MSC device there is also a way to insert and remove the media (without a media your device will be represented in the system as a SD card reader without a card). But for that, if you still want to use esp-idf, you need to use tinyusb component directly, as to provide media you need to use `bool tud_msc_test_unit_ready_cb(uint8_t lun)` and return `true` in this callback.
In suvh case you will have both classes all the time, but the media in MSC class will be available (as well and READ/WRITE SCSI commands) only when you need it. 

Otherwise, there is a version of `tusb_teardown()` implementation awaiting for the review in here: https://github.com/hathach/tinyusb/pull/2904

So, there are several options to try it out. Maybe it will help you with your idea.

marchingband, 2025-01-02:
@roma-jam thank you so much for your help. 
Using `tud_msc_test_unit_ready_cb` and friends worked well.
I had to comment-out `tud_msc_test_unit_ready_cb` from `tusb_msc_storage.c`, so I am no longer in sync with the repo, but that's ok, it works :)

marchingband, 2025-01-05:
There is one bug(?) where the driver does not respect `tud_msc_test_unit_ready_cb` at initialization time.
I can set it to always return `false` and it still takes the drive. This only happens once, during init, later it respects `tud_msc_test_unit_ready_cb`.
I setup a workaround to just always unmount from the MSC following initialization, but I think it will be pretty common for user to want to initialize the USB driver at boot (because they are doing a composite device for example), but not mount the drive to the MSC.
My workaround is fine for me, just reporting this incase you want to look into it.

marchingband, 2025-01-05:
just realized this may simply be because of
```
// // Invoked when device is mounted (configured)
void tud_mount_cb(void)
{
    tinyusb_msc_storage_unmount();
}
```
In which case just ignore me heh.

hathach, 2025-02-07:
check out `examples/device/dynamic_configuration`, this can be done by dynamically change the configuration descriptor, and tud_disconect() is needed to force host to re-enumerate (and VID/PID maybe required to prevent host caching config/driver)

closed 2025-02-07 by hathach

inbarab, 2025-02-09:
Hi @marchingband,
Question: are you using the CDC as a CLI or just for output?
Thanks
Inbar

marchingband, 2025-02-09:
@inbarab i was using it for input and output, not for a CLI but for a serial interface for a web app actually.
I ended up abandoning this approach and have both CDC and MSC at the same time.

inbarab, 2025-02-10:
When I try to use the tiny usb cdc-acm for input and output (in my case, the esp-idf cli), the I get a task watch dog on the repl task (see stack dump below)
Any ideas what I'm doing wrong?
I took the esp-idf example of tusb console and added the esp-idf console, I can upload the changes I made 

PS C:\dev\ESP\tusb_console> xtensa-esp32s3-elf-addr2line -pfiaC -e build/tusb_console.elf 0x42014FAF:0x3FC93CB0 0x420153CC:0x3FC93CD0 0x403770E9:0x3FC93D00 0x400559DD:0x3FC9CC50 0x4037D00E:0x3FC9CC60 0x4037AE7D:0x3FC9CC80 0x4200CEA5:0x3FC9CCC0 0x4200CFBB:0x3FC9CCE0 0x4200C44F:0x3FC9CD00 0x4200B55D:0x3FC9CD20 0x4200BA1E:0x3FC9CD50 0x42006B4E:0x3FC9CD70 0x4201E8EE:0x3FC9CD90 0x420098D9:0x3FC9CDB0 0x42009CE9:0x3FC9CDE0 0x4200A014:0x3FC9CE30 0x4200A051:0x3FC9CE50 0x42008D84:0x3FC9CE70 0x4037B571:0x3FC9CEA0

0x42014faf: task_wdt_timeout_handling at C:/dev/ESP/esp-idf523/components/esp_system/task_wdt/task_wdt.c:441
0x420153cc: task_wdt_isr at C:/dev/ESP/esp-idf523/components/esp_system/task_wdt/task_wdt.c:515
0x403770e9: _xt_lowint1 at C:/dev/ESP/esp-idf523/components/xtensa/xtensa_vectors.S:1240
0x400559dd: ?? ??:0
0x4037d00e: vPortClearInterruptMaskFromISR at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
 (inlined by) xTaskGetSchedulerState at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/tasks.c:4982
0x4037ae7d: xQueueGenericSend at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/queue.c:941
0x4200cea5: osal_mutex_unlock at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:171
 (inlined by) _ff_unlock at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:48
 (inlined by) _tu_fifo_write_n at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:551
0x4200cfbb: tu_fifo_write_n at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:863
0x4200c44f: tud_cdc_n_write at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/class/cdc/cdc_device.c:178 (discriminator 4)
0x4200b55d: tud_cdc_n_write_char at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/class/cdc/cdc_device.h:102
 (inlined by) tinyusb_cdcacm_write_queue_char at C:/dev/ESP/tusb_console/managed_components/espressif__esp_tinyusb/tusb_cdc_acm.c:235
0x4200ba1e: tusb_write at C:/dev/ESP/tusb_console/managed_components/espressif__esp_tinyusb/vfs_tinyusb.c:127
0x42006b4e: esp_vfs_write at C:/dev/ESP/esp-idf523/components/vfs/vfs.c:482 (discriminator 4)
0x4201e8ee: write at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11 (discriminator 1)
0x420098d9: linenoiseInsertPastedChar at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:698
0x42009ce9: linenoiseEdit at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:892
0x4200a014: linenoiseRaw at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:1105
0x4200a051: linenoise at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:1168
0x42008d84: esp_console_repl_task at C:/dev/ESP/esp-idf523/components/console/esp_console_repl.c:530
0x4037b571: vPortTaskWrapper at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

marchingband, 2025-02-10:
@inbarab This isn't a good place to get support for this error, it's almost certainly unrelated to the driver. You could post your code on Reddit or some beginner friendly forum.

roma-jam, 2025-06-13:
Hi @marchingband,

there is a big refactoring in `esp_tinyusb` component on the way: https://github.com/espressif/esp-usb/pull/178

It will unlock the possibility to define user behavior on the tud mount/unmount callbacks as well as use the previous default internal logic (mount/umount on the callbacks).

With the already available option to re-install tinyusb driver from `esp_tinyusb` component, this could help you to achieve the initial goal:  

> it needs to change into a MSC device for a while, and then back to serial again.