hathach/tinyusb #2927: dwc2_dcache_clean() missing for STM32H7 build

MWP, 2025-01-02:
### Operating System

Others

### Board

Custom

### Firmware

Custom STM32H7A3 based device build.

### What happened ?

When CFG_TUD_MEM_DCACHE_ENABLE is set, dwc2_dcache_clean() function is missing.

### How to reproduce ?

Build a project using TinyUSB for STM32H7xx with CFG_TUD_MEM_DCACHE_ENABLE set.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-01-02:
Please post your build log.

MWP, 2025-01-02:
I cannot :(

It's pretty clear though.
It seems the data cache handling functions were added for the esp32 target, but were not for the stm32.

A quick fix for the problem is adding the following to the end of dwc2_stm32.h
```
TU_ATTR_ALWAYS_INLINE static inline bool dwc2_dcache_clean(const void* addr, uint32_t data_size) {
  data_size = round_up_to_cache_line_size(data_size);
  SCB_CleanDCache_by_Addr((uint32_t*)addr, data_size);
  return true;
}

TU_ATTR_ALWAYS_INLINE static inline bool dwc2_dcache_invalidate(const void* addr, uint32_t data_size) {
  data_size = round_up_to_cache_line_size(data_size);
  SCB_InvalidateDCache_by_Addr((uint32_t*)addr, data_size);
  return true;
}

TU_ATTR_ALWAYS_INLINE static inline bool dwc2_dcache_clean_invalidate(const void* addr, uint32_t data_size) {
  SCB_CleanInvalidateDCache_by_Addr((uint32_t*)addr, data_size);
  return true;
}
```

closed 2025-08-08 by hathach