hathach/tinyusb #2931: AUDIO - Flow control not working with only 1 supported sample-rate

gmikitiuk, 2025-01-06:
### Operating System

Windows 10

### Board

Custom design

### Firmware

Modified uac2_headset example

### What happened ?

I've noticed that whenever only one sampling frequency is available in `sample_rates`, Windows stops asking for current frequency and this is the only way(to my knowledge) to get `sample_rate_tx` initialized which leads to no flow control working(due to uninitialized `sample_rate_tx`.

### How to reproduce ?

Configure a single frequency in `sample_rates`.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

No logs possible with my host

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-01-06:
Never tried this case, Windows neither get nor set sample rate ?

gmikitiuk, 2025-01-07:
I didn't check it, as in my use-case sample rate is fixed, so it's marked as read only in descriptor. I don't even expect a set call.
What is sure is that get is never called if only one sample rate is available(I guess it assumes that it must be the one presented in the list).

I've worked around it by providing two sample rates but no ability to change it.

HiFiPhile, 2025-02-08:
I took a look and couldn't reproduce the issue with `audio_4_channel_mic` example where sample rate is fixed and marked as read only in descriptor.

When sample rate is read only Windows won't call set sample rate but get sample rate is always called.