hathach/tinyusb #2934: tusb_config.h mising in /pico-sdk/lib/tinyusb/src/

g3noce, 2025-01-08:
### Operating System

Linux

### Board

pico

### Firmware

don't know

### What happened ?

/pico-sdk/lib/tinyusb/src/tusb_option.h:238:12: fatal error: tusb_config.h: No such file or directory
  238 |   #include "tusb_config.h"

### How to reproduce ?

i include this:
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/gpio.h"
#include "bsp/board.h"
#include "tusb.h"

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

In file included from /mnt/e/Genoce/pico/pico-sdk/lib/tinyusb/src/common/tusb_common.h:74,
                 from /mnt/e/Genoce/pico/pico-sdk/lib/tinyusb/src/tusb.h:37,
                 from /mnt/e/Genoce/pico/pico-sdk/lib/tinyusb/hw/bsp/board_api.h:39,
                 from /mnt/e/Genoce/pico/pico-sdk/src/rp2_common/tinyusb/include/bsp/board.h:10,
                 from /mnt/e/Genoce/pico/external_mouse/external_mouse.c:4:
/mnt/e/Genoce/pico/pico-sdk/lib/tinyusb/src/tusb_option.h:238:12: fatal error: tusb_config.h: No such file or directory
  238 |   #include "tusb_config.h"
      |            ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/external_mouse.dir/build.make:79: CMakeFiles/external_mouse.dir/external_mouse.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2257: CMakeFiles/external_mouse.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

OrhanYigitDurmaz, 2025-01-17:
> ### Operating System
>
> Linux
> ### Board
>
> pico
> ### Firmware
>
> don't know
> ### What happened ?
>
> /pico-sdk/lib/tinyusb/src/tusb_option.h:238:12: fatal error: tusb_config.h: No such file or directory 238 | [...]

you need to create a tusb_config.h file

dhalbert, 2025-01-17:
Look here: `pico-sdk/src/rp2_common/pico_stdio_usb/include/tusb_config.h`

hathach, 2025-02-07:
tusb_config.h must be provided by application not the stack itself

closed 2025-02-07 by hathach