hathach/tinyusb #2942: Make inclusion of stdio.h optional

chintal, 2025-01-12:
### Related area

common includes

### Hardware specification

hardware independent

### Is your feature request related to a problem?

I'm using TinyUSB with a custom embedded stack which implements its own printf and associated functions.

TinyUSB includes stdio.h in src/common/tusb_common.h. This results in multiple declarations of sprintf(), which results in compile time errors. 

It is unclear to me which parts, if any, of stdio.h are used by tinyusb. With the basic setup of tinyUSB I have in place, commenting out the include seems to cause no apparent breakage at compile time (though this is not tested in any meaningful way yet).

### Describe the solution you'd like

What I would like to have is a way to disable the inclusion of stdio.h by tinyusb by changing something in the configuration. If the code does use something from stdio.h, then by setting this configuration, I would either be accepting that 
  - those features (explicitly listed somewhere) are no longer available to me, or
  - i would have to provide these necessary components from stdio.h (also listed somewhere) to get those features back

I am commenting out this inclusion for the moment, though I would like to avoid doing this in favor of being able to use vanilla tinyUSB, appropriately configured through its configuration files.   

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

fenugrec, 2025-02-22:
Quick glance, looks like
- `src/common/tusb_debug.h` uses printf and snprintf;
- hw/mcu/sony/cxd56 has some fseek calls but it also includes `stdio` itself so not an issue
- `NULL` is also defined in stddef which is included anyway

running `include-what-you-use` after removing the stdio.h include might provide more clues.

torrent233, 2025-05-15:
I looked into `src/common/tusb_common.h` and the use of `stdio.h`.

It seems that `stdio.h` is included unconditionally, but from reviewing the codebase, most of TinyUSB's core functionality does not actually depend on it. The only potential usages appear to be in debugging macros (e.g., `TU_ASSERT`, `TU_VERIFY`) which rely on `tu_printf()` when `CFG_TUSB_DEBUG` is enabled.

To improve compatibility with custom embedded environments — especially those providing their own lightweight `printf()` — I suggest making the inclusion of `stdio.h` configurable.

For example, we could wrap the include statement with a macro:

```c
#ifndef TINYUSB_DISABLE_STDIO
#include <stdio.h>
#endif
```

closed 2025-11-05 by hathach