hathach/tinyusb #2993: ESP32-S3 dwc2 driver broken, FIFO size limited to 1/4

dzingoni, 2025-02-18:
### Operating System

MacOS

### Board

ESP32-s3 devkit

### Firmware

Any firmware using ESP32-s3 usb device, I worked on usb audio.

### What happened ?

Look like there is a bug in calculation of available FIFO RAM for esp32-s3 (and also for P4 as I can see).

In file dwc2_esp32.h we can find this definition:
static const dwc2_controller_t _dwc2_controller[] = {
  { .reg_base = DWC2_FS_REG_BASE, .irqnum = ETS_USB_INTR_SOURCE, .ep_count = 7, .ep_in_count = 5, **.ep_fifo_size = 1024** }
};

where we see that ep_fifo_size is 1024. **These are words, not bytes**, as far as I know.

Now when the dwc2 driver uses this number in the routine (file dcd_dwc2.c):

static void dfifo_device_init(uint8_t rhport) {
  const dwc2_controller_t* dwc2_controller = &_dwc2_controller[rhport];
  dwc2_regs_t* dwc2 = DWC2_REG(rhport);
  dwc2->grxfsiz = calc_device_grxfsiz(CFG_TUD_ENDPOINT0_SIZE, dwc2_controller->ep_count);

  // Scatter/Gather DMA mode is not yet supported. Buffer DMA only need 1 words per endpoint direction
  const bool is_dma = dma_device_enabled(dwc2);
 **_dcd_data.dfifo_top = dwc2_controller->ep_fifo_size/4;**
  if (is_dma) {

...
we can see that **ep_fifo_size is divided by 4**, and so _dcd_data.dfifo_top is 256 (words) and not 1024.

Now with 256 words it's impossible to go over 16 bit stereo 48000Hz with IN and OUT. 32bit is impossible and generates an ASSERT.
Even 24 bit stereo playback only at 96Khz is impossible.

From what I know the 1024 ep_fifo_size is in WORDS, not BYTES. So we should either change it to 4096 (in bytes) or remove the /4 division. 

### How to reproduce ?

Just try to run the uac2_headset example and set it to 48Khz, 32 bit x sample, stereo (2 channels out, 2 channels in).

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

  No need to debug, to see the asserts just set the debugg level to 3 in tusb_config.h

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-02-18:
value is already correct

closed 2025-02-18 by hathach

HiFiPhile, 2025-02-18:
> if we consider to have only 256 words and not 1024 for buffers on the esp32s3 the usb is very limited.

No, ESP32-S3 has only 256 word or 1024 bytes for USB FIFO, the value 4096 is for STM32 who has a bigger FIFO.

From reference manual:

[image]

dzingoni, 2025-02-18:
OK, thank you , I had seen this ... and it's incoerent with the image in the technical reference ...

[image]

Moreover I was examing the fork of Tinyusb by expressif ... 
https://github.com/espressif/tinyusb/blob/82a7a644ccbe5a38c725517ca3baf0703f90b2fb/src/portable/espressif/esp32sx/dcd_esp32sx.c

I don't have access to more detailed info, but trying to understand the code and reasoning on everything I believe that 1024 is the limit for each FIFO, and not global. From the image I'd bet also that different endpoints should have fifos at specific locations in the memory map (1000, 2000 and so on ) ... but again, I miss any other technical doc.