hathach/tinyusb #3004: issues in examples directory prevent pico-examples project from building

rppicomidi, 2025-02-28:
### Operating System

Linux

### Board

Raspberry Pi Pico

### Firmware

[pico-examples](https://github.com/raspberrypi/pico-examples)

### What happened ?

I tried to use the [pico-examples](https://github.com/raspberrypi/pico-examples) project in the VS Code IDE. The project does not build. Refer to pull request #1627.

### How to reproduce ?

1. Start with tinyusb code after commit 1cfc88d
2. Clone the [pico-examples](https://github.com/raspberrypi/pico-examples) project
3. Install Microsoft VS Code IDE per the [instructions](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)
4. Open a Linux terminal window
5. `export PICO_SDK_PATH=[path to the pico-sdk directory]`
6. `export PICO_BOARD=pico`
7. `export BOARD=raspberry_pi_pico`
8. `export FAMILY=rp2040`
9. Start VS Code by typing the command `code`
10. File->Open Folder... and open the `pico-examples` project directory where you cloned it
11. Set the toolchain, etc.
12. Observe the project fails to build because the board_test, msc_dual_lun, and and cdc_msc `CMakeLists.txt` files cannot execute the `family_configure_device_example()` call.

If I replace the last line of each `CMakeLists.txt` with `family_configure_device_example(${EXE_NAME} noos)` then this problem does not happen.

However, when I try to build any tinyusb example by choosing the project in the CMake extension tool, then the example fails to build because `family.c` cannot find `board.h`.  If I copy the `board.h` file from the `raspberry_pi_pico` directory to the `pico-sdk` directory, then the problem is fixed. Adding an empty board.h file here also works.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-02-28:
I don't use vscode, can you provide equivalent cmake command to test with.

rppicomidi, 2025-02-28:
I am sorry to argue. I do not love VS Code IDE either. However, the Raspberry Pi documentation instructs most of their users to use VS Code. The average Raspberry Pi Pico C/C++ application developer pool will encounter this issue as I described it. It is good to try Raspberry Pi Pico example code using VS Code. Same reason I try the Arduino IDE, too.

But from the command line, to build the device_info example:
```
export PICO_SDK_PATH=[your path to the pico-sdk]
export PICO_BOARD=pico
export PICO_PLATFORM=rp2040
git clone https://github.com/raspberrypi/pico-examples.git
cd pico-examples
mkdir build
cd build
cmake -DBOARD='raspberry_pi_pico' -DFAMILY='rp2040' ..
cd usb/host/tinyusb_host_examples/device_info
```
If you do not make the changes to board_test, msc_dual_lun, and and cdc_msc `CMakeLists.txt` files, the `cmake` step fails. If you do not provide a `board.h` file to the `boards/pico-sdk` directory, then then `make` step fails when it finally tries to build tinyusb examples.

hathach, 2025-02-28:
ah thanks, seems like it is more than board.h in pico-sdk. I may make more changes as well, let me reproduce and try to fix it

hathach, 2025-03-05:
@rppicomidi which changes you need to make to the CMakeLists.txt. I have tried but it seems to compile the pico-examples just fine with current master (already added board.h to bsp/rp2040/pico_sdk)

```
cmake -DBOARD='raspberry_pi_pico' -DFAMILY='rp2040' ..
Using PICO_SDK_PATH from environment ('/home/hathach/code/pico/pico-sdk')
PICO_SDK_PATH is /home/hathach/code/pico/pico-sdk
Initializing PICO_BOARD from environment ('pico')
Initializing PICO_PLATFORM from environment ('rp2040')
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/hathach/code/pico/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_EXTRAS_PATH as sibling of PICO_SDK_PATH: /home/hathach/code/pico/pico-sdk/../pico-extras
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/hathach/app/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/hathach/app/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /home/hathach/app/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc
Build type is Release
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter 
TinyUSB available at /home/hathach/code/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/hathach/code/pico/pico-sdk/lib/btstack
cyw43-driver available at /home/hathach/code/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/hathach/code/pico/pico-sdk/lib/lwip
mbedtls available at /home/hathach/code/pico/pico-sdk/lib/mbedtls
Only building blink_any for non W boards as PICO_CYW43_SUPPORTED is not set
Skipping encrypted example which is unsupported on this platform
Skipping hello_dcp example which is unsupported on this platform
Skipping runtime_flash_permissions example which is unsupported on this platform
Skipping dvi_out_hstx_encoder example which is unsupported on this platform
Skipping spi_lcd example which is unsupported on this platform
Skipping multicore_doorbell example which is unsupported on this platform
Skipping hello_otp example which is unsupported on this platform
Skipping boot_info example which is unsupported on this platform
Skipping universal examples as PICO_RISCV_TOOLCHAIN_PATH and PICO_ARM_TOOLCHAIN_PATH are not defined
Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable
Skipping SHA256 examples as pico_sha256 or pico_mbedtls unavailable
Skipping FreeRTOS examples as FREERTOS_KERNEL_PATH not defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hathach/code/pico/pico-examples/build
```

rppicomidi, 2025-03-09:
Still seems to be an issue. I have installed the Pico-PIO-USB project using the script.

```
~/projects/pico/pico-sdk$ git status
HEAD detached at 2.1.0
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   lib/btstack (new commits)
	modified:   lib/tinyusb (new commits, untracked content)

no changes added to commit (use "git add" and/or "git commit -a")

~/projects/pico/pico-sdk/lib/tinyusb$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	hw/mcu/raspberry_pi/
	lib/FreeRTOS-Kernel/
	lib/lwip/
	tools/uf2/

~/projects/pico/pico-sdk/lib/tinyusb$ git log -1
commit 29ffd57237554b1f2339af543e3789ae04d3b29b (HEAD -> master, origin/master, origin/HEAD)
Merge: 334ac8072 aa860041b
Author: HiFiPhile <admin@hifiphile.com>
Date:   Sat Mar 8 11:57:41 2025 +0100

    Merge pull request #2990 from dauc/master
    
    Add full/high speed compile flag for Microchip SAME70 examples

~/projects/pico/pico-examples$ git log -1
commit 84e8d489ca321a4be90ee49e36dc29e5c645da08 (HEAD -> master, tag: sdk-2.1.1, origin/master, origin/HEAD)
Author: graham sanderson <graham.sanderson@raspberrypi.com>
Date:   Tue Feb 18 19:23:24 2025 -0600

    fix CI on windows

~/projects/pico/pico-examples$ rm -rf build
~/projects/pico/pico-examples$ mkdir build
~/projects/pico/pico-examples$ cd build
~/projects/pico/pico-examples/build$ export PICO_SDK_PATH=${HOME}/projects/pico/pico-sdk
~/projects/pico/pico-examples/build$ export PICO_BOARD=pico
~/projects/pico/pico-examples/build$ export PICO_PLATFORM=rp2040

~/projects/pico/pico-examples/build$ cmake -DBOARD='raspberry_pi_pico' -DFAMILY='rp2040' ..
Using PICO_SDK_PATH from environment ('/home/paul/projects/pico/pico-sdk')
PICO_SDK_PATH is /home/paul/projects/pico/pico-sdk
Initializing PICO_BOARD from environment ('pico')
Initializing PICO_PLATFORM from environment ('rp2040')
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/paul/projects/pico/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter 
TinyUSB available at /home/paul/projects/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/paul/projects/pico/pico-sdk/lib/btstack
cyw43-driver available at /home/paul/projects/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/paul/projects/pico/pico-sdk/lib/lwip
mbedtls available at /home/paul/projects/pico/pico-sdk/lib/mbedtls
Only building blink_any for non W boards as PICO_CYW43_SUPPORTED is not set
Skipping encrypted example which is unsupported on this platform
Skipping hello_dcp example which is unsupported on this platform
Skipping runtime_flash_permissions example which is unsupported on this platform
Skipping dvi_out_hstx_encoder example which is unsupported on this platform
Skipping spi_lcd example which is unsupported on this platform
Skipping multicore_doorbell example which is unsupported on this platform
Skipping hello_otp example which is unsupported on this platform
Skipping boot_info example which is unsupported on this platform
Skipping universal examples as PICO_RISCV_TOOLCHAIN_PATH and PICO_ARM_TOOLCHAIN_PATH are not defined
CMake Error at /home/paul/projects/pico/pico-sdk/lib/tinyusb/examples/device/board_test/CMakeLists.txt:37 (family_configure_device_example):
  family_configure_device_example Function invoked with incorrect arguments
  for function named: family_configure_device_example

CMake Error at /home/paul/projects/pico/pico-sdk/lib/tinyusb/examples/device/cdc_msc/CMakeLists.txt:40 (family_configure_device_example):
  family_configure_device_example Function invoked with incorrect arguments
  for function named: family_configure_device_example

CMake Error at /home/paul/projects/pico/pico-sdk/lib/tinyusb/examples/device/msc_dual_lun/CMakeLists.txt:39 (family_configure_device_example):
  family_configure_device_example Function invoked with incorrect arguments
  for function named: family_configure_device_example

Skipping SHA256 examples as pico_sha256 or pico_mbedtls unavailable
Skipping FreeRTOS examples as FREERTOS_KERNEL_PATH not defined
-- Configuring incomplete, errors occurred!

```

hathach, 2025-03-10:
thank you, I am able to reproduce the issue, this should be fixed by #3019

PR #3005 (2025-02-28): add missing board.h for BOARD=pico_sdk
**Describe the PR**
fix #3004
PR #3019 (2025-03-10): fix(rp2040) set RTOS as cmake cache to fix pico-example build
fix #3004 set RTOS as cmake cache to fix pico-example build
closed 2025-03-10 by hathach