hathach/tinyusb #3016: Could not compile examples

screwdriver99, 2025-03-08:
### Operating System

MacOS

### Board

stm32l412nucleo

### Firmware

examples/device/hid_composite

### What happened ?

Not able to compile examples

### How to reproduce ?

1. Clone the repo
2. Checkout to master branch
3. cd /examples/device/hid_composite
4. add CMSIS device drivers to the inclusion path and build files
5. make board=stm32l412nucleo SPEED=full all

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[shell]$ make board=stm32l412nucleo SPEED=full all
make: *** tinyusb/hw/bsp/broadcom_32bit/boards/: Is a directory.  Stop.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-03-08:
Please remove `all` from `make`

screwdriver99, 2025-03-08:
> Please remove `all` from `make`

The problem remains

HiFiPhile, 2025-03-08:
Ops, `BOARD` should be in upper case,  also since stm32l412 is FS only so need to add `SPEED`: `make BOARD=stm32l412nucleo all`

closed 2025-03-08 by HiFiPhile