hathach/tinyusb #3022: High speed support to MXP's iMXRT family USB host

ThomasWongNZ, 2025-03-11:
### Operating System

Others

### Board

iMXRT1052EVB

### Firmware

Currently high speed is disabled for iMXRT family CPUs USB host, as mentioned in the comment:
https://github.com/hathach/tinyusb/blob/master/src/portable/chipidea/ci_hs/hcd_ci_hs.c#L99

When will it be fixed?

### What happened ?

When the host in high speed:
1. Have issue when plug/unplug devices.
2. Does not detect disconnection

### How to reproduce ?

1. Override full speed mode by commenting out statement
https://github.com/hathach/tinyusb/blob/master/src/portable/chipidea/ci_hs/hcd_ci_hs.c#L99
2. Build examples/host/msc_file_explorer
3. Run complied binary with iMXRT1052EVB

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[tinyusb-debug-log.txt](https://github.com/user-attachments/files/19198929/tinyusb-debug-log.txt)

### Screenshots

Device unplug not detected.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-03-11:
Please don't repost known issue

closed 2025-03-11 by HiFiPhile

ThomasWongNZ, 2025-03-12:
Sorry, where is the socalled known issue?

hathach, 2025-03-19:
HiFiPhile means this issue is known by maintainer and we just don't have time and motivation to address it atm. Regarding the question "When will this be fixed ?" the answer is always "when someone makes an PR"

hathach, 2025-04-29:
should be fixed by https://github.com/hathach/tinyusb/pull/3104