hathach/tinyusb #3026: Audio control transfers misaligned on some platforms

sodank808, 2025-03-12:
### Operating System

Others

### Board

ESP32-S3

### Firmware

Probably the most relevant example is uac2_headset.

### What happened ?

After the 0.18.0 release, certain platforms such as iOS fail to record the microphone because the endpoint stalls. This was root caused to missing 4 byte alignment on ctrl_buf_1. It seems that this changed in #2865 but it is unclear why it was changed, probably a mistake.

### How to reproduce ?

Record microphone in a USB audio class 2.0 example. The original replication was on iOS in voice memo. Unfortunately, alignment issues can be finicky and depend on the arrangement of memory performed by the linker.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Unfortunately, the device under test does not have logging piped out.

### Screenshots

In the USB analyzer, I was able to clue in to the issue when the IN endpoint stopped sending data right after a control transfer

[image]

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-03-13:
Hi, with a first look I think the issue should be elsewhere, since with DWC2 IP (used by ESP32) there isn't an alignment requirement for control transfer buffer. Also data buffer aligment is only needed when DMA is enabled.

_Meanwhile ctrl_buf_1 should be aligned for some NXP MCUs_

closed 2025-03-17 by HiFiPhile

HiFiPhile, 2025-10-04:
@sodank808 Hi, I came back as I'm doing UAC1 support in #3270. After examining this part of code, there is no reason `ctrl_buf_1` needs a alignment requirement. Also a control transfer in endpoint 0 won't affect data endpoints.