hathach/tinyusb #3030: Latest usbh.c changes "breaks" some devices

Cpasjuste, 2025-03-14:
### Operating System

Linux

### Board

rp2040

### Firmware

https://github.com/Cpasjuste/uGamePad

### What happened ?

Hi,

In my (rp2040/rp2350) project, i did update tinyusb to be up to date. But since then, i have a device which is not working anymore in host mode. This is the "Neo Geo Mini USB-C Gamepad". 

The problem comes from the addition of the "newly added" enumerations in [usbh.c](https://github.com/hathach/tinyusb/blob/5333d042f9384ec1cc0663a2708c0ca7cb5cdc32/src/host/usbh.c#L1527):

- ENUM_GET_STRING_LANGUAGE_ID
- ENUM_GET_STRING_MANUFACTURER
- ENUM_GET_STRING_PRODUCT
- ENUM_GET_STRING_SERIAL

In fact, the problem does not comes from tinyusb, but from the vendor i guess, since the queries just hang forever (no timeout). I'm not sure if tinyusb should handle this strings itself, maybe this should be handled by the user if needed ?

Else the root of the problem is probably, as [mentioned in the code](https://github.com/hathach/tinyusb/blob/5333d042f9384ec1cc0663a2708c0ca7cb5cdc32/src/host/usbh.c#L646), the lack of timeout on (tuh_control_xfer)[https://github.com/hathach/tinyusb/blob/5333d042f9384ec1cc0663a2708c0ca7cb5cdc32/src/host/usbh.c#L647C6-L647C22]. 

I have "fixed" (crappy fix) this on my side, but i'm not sure i'll have the time to make a proper fix. 

Thanks,
Cpasjuste

### How to reproduce ?

Hard to reproduce without a crappy device descriptor like the NeoGeo Mini :)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

No debug needed i guess !

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-03-19:
Those are standard request, I dont see why it could cause the issue, can you get and post the debug log and the device info using pc such as lsusb -v here

Cpasjuste, 2025-03-19:
Hi,

Yes I agree, this is strange. The device is a proprietary thing thought, maybe it's not "compliant".  

I would need to buy a usb-c to usb-a someday (wanted to do that since a while), so for I now I will try to take a closer look on the pico in the next days and report back.

Thanks !

hathach, 2025-05-21:
there is moe update, please pull and test again. let me know if it works for you