hathach/tinyusb #3031: Add a get_daddr(idx) function to the new midi_host driver

rppicomidi, 2025-03-14:
### Related area

The midi_host driver addresses the attached device by index. Device string get functions need the device address.

### Hardware specification

Is hardware independent. I am testing on Raspbery Pi Pico

### Is your feature request related to a problem?

I want to be able to fetch Manufacturer, Product and other device strings without blocking the code. The host API that supports that requires the device address. The mount callback returns the device address within the callback data, but now I have 2 values to track: the device index and the device address. An application can do this, but it should only need to track the device index.

### Describe the solution you'd like

I would like a function in `midi_host.c` that returns the device address given the device index provided in the mount callback. Perhaps it needs in the future another function that returns the interface given the device index. That way, the application does not need to always track two numbers for each attached device.

Possible API:
uint8_t tuh_midi_get_daddr_from_idx(uint8_t idx);
uint8_t tuh_midi_get_intf_from_idx(uint8_t idx);

@hathach if you will approve this API, I can write a pull request.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-03-15:
we can implement tuh_cdc_itf_get_info(uint8_t idx, tuh_itf_info_t* info); similar to https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_host.c#L287

rppicomidi, 2025-03-15:
OK. Pull request coming shortly.

closed 2025-03-20 by rppicomidi