hathach/tinyusb #3035: Cmake default generator

Guilty13, 2025-03-16:
### Operating System

Linux

### Board

*

### Firmware

.

### What happened ?

Seems like cmake on my system was installed by Visual Studio, and default generator was Visual Studio 17. As a person not really familiar with cmake it took some time to understand it and fix it with env variables. Maybe it make sense to enforce needed generator to avoid this problem?

### How to reproduce ?

.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-03-16:
You can use CMake Presets: `cmake --preset lpcxpresso55s69`, `cmake --build --preset lpcxpresso55s69` 

> Maybe it make sense to enforce needed generator to avoid this problem?

It's impossible with `CMakeLists.txt`

closed 2025-03-16 by HiFiPhile

Guilty13, 2025-03-17:
How it is intended to do when building from make?

hathach, 2025-03-17:
> How it is intended to be done when building from make? (As I can see, make calls cmake)

I will update the readme later, it basically, you can pick either make/ninja as generator

```
cmake -Bbuild_dir BOARD=adafruit_feather_nrf52840
cmake -Bbuild --build
```

Guilty13, 2025-03-17:
So instead of 

`make BOARD=raspberry_pi_pico FAMILY=rp2040 CMAKE_DEFSYM='-G "Unix Makefiles" ' all`
 should I use 

`cmake -Bbuild_dir BOARD=raspberry_pi_pico FAMILY=rp2040
cmake -Bbuild_dir --build`

(Not at home rn, I can test command above in 9 hours)