hathach/tinyusb #3037: Cannot clone the pio_usb folder for raspberry pi pico

hexad, 2025-03-16:
### Operating System

Linux

### Board

pico

### Firmware

I am trying to build a project that uses this library.
yet it constantly fails.
I do not have the `dcd_pio_usb.c` and `hcd_pio_usb.c` files. I am missing the `pio_usb` directory (inside `~/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi` there is only an `rp2040` folder).
I tried to manually create the folder and put the files in manually, from this repo, however, the build still fails.

I am suspecting I have an older version of tinyusb.
I tried to reinstall with:

```
cd ~/pico/pico-sdk
rm -rf tinyusb
git submodule update --init --recursive
```

However, it still does NOT have the `pio_usb` directory is still missing. What do I have to do to install the latest version of `tinyUSB`? I have spent so much time on trying to build a simplevgithub repo and I am pulling my hair out. Thanks!

### What happened ?

When copying the files from this repo, `cmake ..` passed, but `make` failed with lots of errors.

### How to reproduce ?

I really really do not know how to reproduce if this involves an older version of tinyusb.
But why does the git command does not install the latest?

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

This is what happens when i type make:

[ 38%] Building C object host_hid_to_device_cdc/CMakeFiles/host_hid_to_device_cdc.dir/host_hid_to_device_cdc.c.obj
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:66:49: error: 'HID_KEYCODE_TO_ASCII' undeclared here (not in a function)
   66 | static uint8_t const keycode2ascii[128][2] =  { HID_KEYCODE_TO_ASCII };
      |                                                 ^~~~~~~~~~~~~~~~~~~~
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c: In function 'core1_main':
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:77:3: warning: implicit declaration of function 'tuh_configure'; did you mean 'clock_configure'? [-Wimplicit-function-declaration]
   77 |   tuh_configure(1, TUH_CFGID_RPI_PIO_USB_CONFIGURATION, &pio_cfg);
      |   ^~~~~~~~~~~~~
      |   clock_configure
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:77:20: error: 'TUH_CFGID_RPI_PIO_USB_CONFIGURATION' undeclared (first use in this function)
   77 |   tuh_configure(1, TUH_CFGID_RPI_PIO_USB_CONFIGURATION, &pio_cfg);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:77:20: note: each undeclared identifier is reported only once for each function it appears in
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:81:3: warning: implicit declaration of function 'tuh_init'; did you mean 'tusb_init'? [-Wimplicit-function-declaration]
   81 |   tuh_init(1);
      |   ^~~~~~~~
      |   tusb_init
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:84:5: warning: implicit declaration of function 'tuh_task' [-Wimplicit-function-declaration]
   84 |     tuh_task(); // tinyusb host task
      |     ^~~~~~~~
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c: In function 'main':
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:100:3: warning: implicit declaration of function 'tud_init'; did you mean 'tusb_init'? [-Wimplicit-function-declaration]
  100 |   tud_init(0);
      |   ^~~~~~~~
      |   tusb_init
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:103:5: warning: implicit declaration of function 'tud_task' [-Wimplicit-function-declaration]
  103 |     tud_task(); // tinyusb device task
      |     ^~~~~~~~
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:104:5: warning: implicit declaration of function 'tud_cdc_write_flush' [-Wimplicit-function-declaration]
  104 |     tud_cdc_write_flush();
      |     ^~~~~~~~~~~~~~~~~~~
[... 61 lines cut ...]
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:258:39: error: 'hid_mouse_report_t' undeclared (first use in this function)
  258 |       process_mouse_report(dev_addr, (hid_mouse_report_t const*) report );
      |                                       ^~~~~~~~~~~~~~~~~~
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:258:57: error: expected ')' before 'const'
  258 |       process_mouse_report(dev_addr, (hid_mouse_report_t const*) report );
      |                                      ~                  ^~~~~~
      |                                                         )
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:258:65: error: expected ')' before 'report'
  258 |       process_mouse_report(dev_addr, (hid_mouse_report_t const*) report );
      |                                                                 ^~~~~~~
      |                                                                 )
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:246:84: warning: unused parameter 'report' [-Wunused-parameter]
  246 | void tuh_hid_report_received_cb(uint8_t dev_addr, uint8_t instance, uint8_t const* report, uint16_t len)
      |                                                                     ~~~~~~~~~~~~~~~^~~~~~
At top level:
/home/user1/Desktop/pico_projects/experiment/Pico-PIO-USB/examples/host_hid_to_device_cdc/host_hid_to_device_cdc.c:66:22: warning: 'keycode2ascii' defined but not used [-Wunused-const-variable=]
   66 | static uint8_t const keycode2ascii[128][2] =  { HID_KEYCODE_TO_ASCII };
      |                      ^~~~~~~~~~~~~
make[2]: *** [host_hid_to_device_cdc/CMakeFiles/host_hid_to_device_cdc.dir/build.make:76: host_hid_to_device_cdc/CMakeFiles/host_hid_to_device_cdc.dir/host_hid_to_device_cdc.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1614: host_hid_to_device_cdc/CMakeFiles/host_hid_to_device_cdc.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

closed 2025-03-17 by hathach

hathach, 2025-03-17:
thanks for the issue, it should be fixed now, I change to use an wip pio-usb fork previously.

hexad, 2025-03-17:
Thank you for the effort, but I tried exactly the same thing:
cd ~/pico/pico-sdk/lib
rm -rf tinyusb
git submodule update --init --recursive

And i get exactly the same results. Do i have to do something differently?
Sorry for the trouble.

hathach, 2025-03-18:
tinyusb use python script instead of sumodule to manage dependencies, try run this in tinyusb/

```
python tools/get_deps.py rp2040
```

hexad, 2025-03-18:
Thanks for the answer. Sorry for annoying you again, but I do not have such a file inside `/tinyusb/`

```
~/pico/pico-sdk/lib/tinyusb/tools$ ls -la
total 52
drwxrwxr-x  4 user1 user1 4096 Μαρ  18 21:20 .
drwxrwxr-x 10 user1 user1 4096 Μαρ  18 21:20 ..
-rw-rw-r--  1 user1 user1 4415 Μαρ  18 21:20 build_board.py
-rw-rw-r--  1 user1 user1 3586 Μαρ  18 21:20 build_esp32sx.py
-rw-rw-r--  1 user1 user1 5456 Μαρ  18 21:20 build_family.py
-rw-rw-r--  1 user1 user1 2321 Μαρ  18 21:20 iar_gen.py
-rw-rw-r--  1 user1 user1 5686 Μαρ  18 21:20 iar_template.ipcf
-rw-rw-r--  1 user1 user1  941 Μαρ  18 21:20 top.mk
drwxrwxr-x  7 user1 user1 4096 Μαρ  18 21:20 uf2
drwxrwxr-x  2 user1 user1 4096 Μαρ  18 21:20 usb_drivers

```

hathach, 2025-03-19:
your pico-sdk and tinyusb is too old, please update both and try again

hexad, 2025-03-19:
Once again, thank you for the comment.
Sorry if I am being a noob, but since I do not have recent experience in this (it's been some time since I played with the pico), and I am afraid of messing things up, what is the reccomended way to update the sdk and the tinyusb library?

I have been trying to update the tinyusb, but I guess this doesn't work since the sdk is old. Does tinyusb come with the sdk?

Is there a guy somewhere?

hathach, 2025-03-19:
check out the rpi pico guide for step by step detail

PR #3038 (2025-03-17): change pio-usb to upstream
**Describe the PR**
change pio-usb to upstream, fix #3037