hathach/tinyusb #3047: freetos crash on esp32-s3 when using option CFG_TUH_DWC2_SLAVE_ENABLE

torsts, 2025-03-23:
### Operating System

Windows 11

### Board

M5stack cardputer with espressif esp32-s3

### Firmware

FREETOS

### What happened ?

I am trying to use cardputer with esp3-s3 microcontroller in USB host mode.
The 1st hurdle is that for ESP32-S3 code for USB PHY initialization is missing in tinyusb. This can be circumvented by using a function from the espressif usb component (usb_new_phy).
After calling usb_new_phy the call to tusb_init will succeed.
tusb_init is followed by tuh_task.
We wait in tuh_task_ext until a device gets plugged in. 
This results in a call to process_enumeration.

When tinyusb gets compiled with CFG_TUH_DWC2_DMA_ENABLE == 1, everything goes according to plan.
Compiling with **CFG_TUH_DWC2_SLAVE_ENABLE** == 1 and running the code, **crashes the OS**.

I could follow the code execution up to inside channel_xfer_start.
With CFG_TUH_DWC2_SLAVE_ENABLE we seem to face kind of interrupt storm from the FIFO that passes execution to handle_txfifo_empty. I did a "blocking wait" at the start of handle_txfifo_empty and got multiple executions of the function while still blocking. This lets me believe, that we get the fifo interrupt multiple times even though we did not yet handle the 1st one. From my understanding this will all happen on the same stack and finally lead to a stack overflow.

As a sidenote: This happens when using synopsys\dwc2 and espressif\esp32sx. It does not happen when using CFG_TUH_DWC2_DMA_ENABLE

### How to reproduce ?

Compile tinyusb for esp32-s3 with option CFG_TUH_DWC2_SLAVE_ENABLE

#define CFG_TUH_ENABLED       1
#define CFG_TUSB_OS                 OPT_OS_FREERTOS
#define CFG_TUH_DWC2_SLAVE_ENABLE 1

Run code that calls tuh_task();

Plugin a USB device and watch the OS crash.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Cannot generate a debug log as the OS crashes and a file cannot be stored a connected microsd card (file size will be 0 as it will not be flushed)

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

torsts, 2025-03-24:
`static usb_phy_handle_t phy_hdl;

/**
 * @brief Main entry point for the ESP-IDF application.
 *
 * Initializes TinyUSB in host mode and continuously polls for USB events.
 * When the device descriptor is received, it is printed to the console.
 */
extern "C" void app_main(void)
{
    // Configure USB PHY
    usb_phy_config_t phy_conf = {
        .controller = USB_PHY_CTRL_OTG,
        .target = USB_PHY_TARGET_INT,
        .otg_mode = USB_OTG_MODE_HOST,
        .otg_speed = USB_PHY_SPEED_UNDEFINED,
        .ext_io_conf = nullptr,
        .otg_io_conf = nullptr
    };

    // init usb phy
    // when not calling this, tusb_init will fail in check_dwc2 due to missing usb clock setup and usb reset
    esp_err_t err = usb_new_phy(&phy_conf, &phy_hdl);

    tusb_rhport_init_t host_init = {
        .role = TUSB_ROLE_HOST,
        .speed = TUSB_SPEED_AUTO
    };

    bool ret = tusb_init(0, &host_init); // initialize host stack on roothub port 0

    // just for debug purpose
    // not required for repro
    setupDisplay();
    M5Display_canvasClear();
    M5Display_printf("debug: usb_new_phy -> %d\n", err);
    M5Display_printf("debug: tusb_init -> %d\n", ret);
    M5Display_canvasPushSprite(0, 0);

    while (true)
    {
        // Process TinyUSB host tasks.
        // this call blocks until it gets an event
        tuh_task();
        // we crashed
        // we never come here when compiling with CFG_TUH_DWC2_SLAVE_ENABLE
`

torsts, 2025-03-24:
@HiFiPhile .. thank you for pointing me to this. If I had known earlier, it would have saved me 2 days of debugging. I took the usb_init code that you pointed me to. It does the same stuff that I figured out myself the hard way.. but the code looks a lot nicer now.

Unfortunately it does not resolve the crash problem with CFG_TUH_DWC2_SLAVE_ENABLE.
I managed to create the debug log with log level 3 now. However, I had to "manually" terminate the logfile in the channel_xfer_start function as otherwise the crash prevents the logfile from being properly closed and written out. There is nothing surprising in there. It looks almost the same as when logging with CFG_TUH_DWC2_DMA_ENABLE. The difference is just the
DMA = 1 / DMA = 0 line and the crash.

Surprisingly CFG_TUD_DWC2_SLAVE_ENABLE for devices seem to work fine. Any idea on how to further pinpoint the problem?

torsts, 2025-03-24:
OK.. don't spent cycles on this. I will try to generate a core dump.

hathach, 2025-03-25:
@torsts can you test the stock example exmaples/host/device_info or cdc_msc_hid_freertos to see if that woks. I personally don't have issue with esp32 mcu. 
@HiFiPhile I will check this later, we can limit NAKs per frame for each endpoint e.g 3 times

torsts, 2025-03-25:
@hathach I will try the examples you are referring to. In addition I will try to increase the stack size. Eventually the stack is more preassured with CFG_TUH_DWC2_SLAVE_ENABLE and the chances for an overflow are bigger.

torsts, 2025-03-25:
Stacksize does not seem to play a role here. I noticed something else. We dont seem to be able to get out of an interrupt signalling state. I did some debugging around hcd_int_handler. We hit GINTSTS_CONIDSTSCHNG, then GINTSTS_HPRTINT and then > 30 x GINTSTS_NPTX_FIFO_EMPTY without ever receiving something. This is strange.. IIRC the 1st packet has 8 bytes max.. so even when sending each byte individually, we should have received something in the meantime. Will have a closer look to handle_txfifo_empty now.

torsts, 2025-03-25:
I tried exmaples/host/device_info. Had to rewrite the code a bit as I dont have a board with LED. I am facing the same behavior. I will keep debugging with my original project.

torsts, 2025-03-25:
Looking deeper.. the OS does not crash. It gets reset with reason ESP_RST_INT_WDT.
Here is the weird thing, we call into handle_txfifo_empty but inside we never call dfifo_write_packet. So the packet does not go onto the wire.
Eventually I am missing some FIFO setup code.

torsts, 2025-03-25:
if ((xact_bytes > (txsts_bm->fifo_available << 2)) || (txsts_bm->req_queue_available == 0)) {
This is why we "fail":
txsts_bm->fifo_available == 0 and txsts_bm->req_queue_available == 0;

hathach, 2025-03-26:
> I tried exmaples/host/device_info. Had to rewrite the code a bit as I dont have a board with LED. I am facing the same behavior. I will keep debugging with my original project.

- you can skip the led, just define a dummy pin. That will make sure we test on the same software (I also prefer to test with stock example as well). 
- Also please provide the LOG=2 output
- Does the issue occurs with other device or only with 1 specific ? what is the device specific make/model and its configurat descriptor.

torsts, 2025-03-26:
Unfortunately I just have 1 device and it is not a developer board. This means: no LEDs, just one USB connector. When USB is used by code, I am loosing the builtin JTAG capability as there is some "hardwiring" to the USB port. I had to add code, that redirects printf to a file that will be stored on a microSD card. In addition I had to close the file before freeRTOS reboots.. otherwise it will have 0 content. I logged with LOG level 3. Here is what I get when using the FIFO (before reset):

USBH init on controller 0, speed = Full
sizeof(usbh_device_t) = 86
sizeof(hcd_event_t) = 12
sizeof(_ctrl_xfer) = 20
sizeof(tuh_xfer_t) = 28
sizeof(tu_fifo_t) = 20
sizeof(tu_edpt_stream_t) = 112
guid, gsnpsid, ghwcfg1, ghwcfg2, ghwcfg3, ghwcfg4
0x00000000, 0x4F54400A, 0x00000000, 0x224DD930, 0x00C804B5, 0xD3F0A030
Fullspeed PHY init
DMA = 0
[0:] USBH Device Attach
Full Speed
[0:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00 

===
In the espressif code I found some  functions called usb_dwc_hal_set_fifo_bias and usb_dwc_ll_hptxfsiz_set_ptx_fifo_size. In addition I read "The underlying hardware has size adjustable FIFOs to cache USB packets.." I will check, how the espressif stack is doing the FIFO stuff.

HiFiPhile, 2025-03-26:
> Unfortunately I just have 1 device and it is not a developer board.

The question is about the devices tested when esp32 is acting as host, like keyboard, usb key, etc.

torsts, 2025-03-26:
I tried 3 different devices and could do more.. the device arrival is recognized but as soon as I want to fetch the 1st descriptor, the bytes from "Get Descriptor" dont go into the fifo.. hence never leave the esp32

roma-jam, 2025-03-26:
I have the same thing on S3 with `device_info` example and default config (`CFG_TUH_DWC2_SLAVE_ENABLE=1`). 

With `CFG_TUH_DWC2_DMA_ENABLE=1` all works. 

Seems that something changed after adding the DMA support. 

Was it working in single mode before adding the DMA?

roma-jam, 2025-03-26:
@torsts Could you try to comment these lines and try again please? 

```
diff --git a/src/portable/synopsys/dwc2/hcd_dwc2.c b/src/portable/synopsys/dwc2/hcd_dwc2.c
index b13479b02..723d1169b 100644
--- a/src/portable/synopsys/dwc2/hcd_dwc2.c
+++ b/src/portable/synopsys/dwc2/hcd_dwc2.c
@@ -814,9 +814,9 @@ static bool handle_txfifo_empty(dwc2_regs_t* dwc2, bool is_periodic) {
 
         // skip if there is not enough space in FIFO and RequestQueue.
         // Packet's last word written to FIFO will trigger a request queue
-        if ((xact_bytes > (txsts_bm->fifo_available << 2)) || (txsts_bm->req_queue_available == 0)) {
-          return true;
-        }
+        // if ((xact_bytes > (txsts_bm->fifo_available << 2)) || (txsts_bm->req_queue_available == 0)) {
+        //   return true;
+        // }
 
         dfifo_write_packet(dwc2, ch_id, edpt->buffer + xfer->fifo_bytes, xact_bytes);
         xfer->fifo_bytes += xact_bytes;
```

hathach, 2025-03-26:
> I have the same thing on S3 with `device_info` example and default config (`CFG_TUH_DWC2_SLAVE_ENABLE=1`).
> 
> With `CFG_TUH_DWC2_DMA_ENABLE=1` all works.
> 
> Seems that something changed after adding the DMA support.
> 
> Was it working in single mode before adding the DMA?

I forgot, I think hcd dwc2 are mostly tested on P4. Slave and DMA are added pretty much same time. I will try to pul out a S3 to test with and maybe added that to the hil pool as well. Btw, does the issue occurs with S2 ?

torsts, 2025-03-26:
Dont have an S2. ;) .. I commented out the lines as requested.. The behavior changes. No watchdog reset anymore. However, I still dont seem to receive the descriptor (at least on the top application level). One thing I can already confirm, with this change I get further and receive a GINTSTS_HCINT -> handle_channel_irq. This is where my current debug centers.

As a sidenote, espressif itself uses the dma in host mode. Will eventually try to force slave mode on them in case I figure out a couple more bits and pieces.

roma-jam, 2025-03-26:
@torsts 

Could you replace the printf with the `esp_rom_printf`, by adding the lines in `tusb_config.h`: 

`
#define CFG_TUSB_DEBUG_PRINTF       esp_rom_printf

`

and then share the log?

torsts, 2025-03-26:
@roma-jam I dont have a developer board.. so just the one (1) builtin USB. From my understanding, jtag and serial print gets lost as soon as I switch the controller into host mode. How exactly would I retrieve the log from esp_rom_printf ?

I in a state where "something (process_enumeration) worked once"... but not stable. Now it feels like a race condition.
I am working on getting a LOG but I likely have to pick a different approach than esp_rom_printf.

torsts, 2025-03-26:
@roma-jam  .. ok.. took a bit .. here the last lines of the log
/--
...
Device configured
Parsing Configuration descriptor (wTotalLength = 32)
Device mounted, address: 1
[0:1] Get Descriptor: 80 06 00 01 00 00 12 00 
on EP 00 with 8 bytes: OK
on EP 80 with 18 bytes: OK
[0:1] Control data:
  0000:  12 01 10 02 00 00 00 40 1C 1B 0E 1A 00 01 02 03  |.......@........|
  0010:  01 01                                            |..|
on EP 00 with 0 bytes: OK

--- Device Descriptor for device 1 ---
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x1b1c
idProduct 0x1a0e
bcdDevice 1.00
iManufacturer 2
iProduct 3
iSerialNumber 1
bNumConfigurations 1
-----------------------------------------
[0:] USBH Device Attach
Full Speed
[0:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00 
on EP 00 with 8 bytes: OK
on EP 80 with 8 bytes: OK
[0:0] Control data:
  0000:  12 01 10 02 00 00 00 40                          |.......@|
on EP 00 with 0 bytes: OK

enum_request_set_addr 1724: ASSERT FAILED
[0:] USBH Device Attach (duplicated)
[0] Aborted transfer on EP 00
Full Speed
[0:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00 
on EP 00 with 8 bytes: OK
on EP 80 with 8 bytes: OK
[0:0] Control data:
  0000:  12 01 01 02 00 00 00 08                          |........|
on EP 00 with 0 bytes: OK

enum_request_set_addr 1724: ASSERT FAILED
--/

This is not the stock sample although pretty close to it. So the removal of the lines you gave me got me up to speed. From the log I can see, that I am missing the "DEVICE REMOVED" event. I will keep digging. Now that the esp32 does not get reset by the watchdog, it is a bit easier to create logs. Btw.. while testing I attached 3 different devices.

torsts, 2025-03-26:
@hathach , @roma-jam , @HiFiPhile  .. I am ready to close this bug report from my side as the original problem seem to be resolved with the removal of the lines. What is your opinion on this?

roma-jam, 2025-03-26:
@torsts I don't think that my suggestion could be a solution, as I just offer that to disable the usage of `txsts_bm` pointer, which uses `gnptxsts` as a source for `fifo_available` value, as this register is not relevant when dedicated FIFO is used. 

But this is not a solution so far as we need another mechanism to handle the available FIFO.

hathach, 2025-03-27:
I am able to reproduce the issue with S3 and host slave mode while it works with dma mode. As discussioned, it is probably an forever blocking while() in the driver. Will try to troubleshoot  this next time I got to work with hcd dwc2. 

```
TinyUSB Device Info Example
[0:] USBH Device Attach
I (346) main_task: Returned from app_main()
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4200c069  PS      : 0x00060034  A0      : 0x8200c0b9  A1      : 0x3fc94d60  
--- 0x4200c069: hcd_int_handler at /home/hathach/code/tinyusb/src/portable/synopsys/dwc2/hcd_dwc2.c:1345

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x8200bfd0  A5      : 0x3fc97ec0  
A6      : 0x00000020  A7      : 0x60080000  A8      : 0x13000034  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x6008002c  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
--- 0x400570e8: memset in ROM
0x400570f3: memset in ROM

Core  0 was running in ISR context:
EPC1    : 0x4201e237  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4200c069
--- 0x4201e237: uart_hal_write_txfifo at /home/hathach/code/esp-idf/components/hal/uart_hal_iram.c:27
0x4200c069: hcd_int_handler at /home/hathach/code/tinyusb/src/portable/synopsys/dwc2/hcd_dwc2.c:1345

Backtrace: 0x4200c066:0x3fc94d60 0x4200c0b6:0x3fc94d80 0x403773d1:0x3fc94da0 0x4200d730:0x3fc97f00 0x4200d8ee:0x3fc97f30 0x4200d911:0x3fc97f80 0x4200d933:0x3fc97fb0 0x4200d084:0x3fc97fe0 0x4200d401:0x3fc98020 0x4200d57d:0x3fc98060 0x42009e76:0x3fc98090 0x4037b945:0x3fc980b0
--- 0x4200c066: hcd_int_handler at /home/hathach/code/tinyusb/src/portable/synopsys/dwc2/hcd_dwc2.c:1342
0x4200c0b6: dwc2_int_handler_wrap at /home/hathach/code/tinyusb/src/portable/synopsys/dwc2/dwc2_esp32.h:78
0x403773d1: _xt_lowint1 at /home/hathach/code/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x4200d730: tuh_control_xfer at /home/hathach/code/tinyusb/src/host/usbh.c:684
0x4200d8ee: _get_descriptor at /home/hathach/code/tinyusb/src/host/usbh.c:1079
0x4200d911: tuh_descriptor_get at /home/hathach/code/tinyusb/src/host/usbh.c:1084
0x4200d933: tuh_descriptor_get_device at /home/hathach/code/tinyusb/src/host/usbh.c:1090
0x4200d084: process_enumeration at /home/hathach/code/tinyusb/src/host/usbh.c:1472
0x4200d401: enum_new_device at /home/hathach/code/tinyusb/src/host/usbh.c:1685
0x4200d57d: tuh_task_ext at /home/hathach/code/tinyusb/src/host/usbh.c:552
0x42009e76: tuh_task at /home/hathach/code/tinyusb/src/host/usbh.h:166
 (inlined by) usb_host_task at /home/hathach/code/tinyusb/examples/host/device_info/src/main.c:289
0x4037b945: vPortTaskWrapper at /home/hathach/code/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Core  1 register dump:
PC      : 0x40378f82  PS      : 0x00060334  A0      : 0x82003085  A1      : 0x3fc9ca80  
--- 0x40378f82: esp_cpu_wait_for_intr at /home/hathach/code/esp-idf/components/esp_hw_support/cpu.c:64

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc9aac0  A5      : 0x3fc9aaa0  
A6      : 0x40375ea0  A7      : 0x00000001  A8      : 0x820116e6  A9      : 0x3fc9ca40  
--- 0x40375ea0: ipc_task at /home/hathach/code/esp-idf/components/esp_system/esp_ipc.c:53

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fc9aaa0  A13     : 0x3fc9aa80  
A14     : 0x00000001  A15     : 0x3fc9cc4c  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40378f7f:0x3fc9ca80 0x42003082:0x3fc9caa0 0x4037c8bd:0x3fc9cac0 0x4037b945:0x3fc9cae0
--- 0x40378f7f: xt_utils_wait_for_intr at /home/hathach/code/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /home/hathach/code/esp-idf/components/esp_hw_support/cpu.c:55
0x42003082: esp_vApplicationIdleHook at /home/hathach/code/esp-idf/components/esp_system/freertos_hooks.c:58
0x4037c8bd: prvIdleTask at /home/hathach/code/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
0x4037b945: vPortTaskWrapper at /home/hathach/code/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

torsts, 2025-03-27:
Took me almost the day.. but looks like I finally figured it out. We seem to face an architectural problem that the compiler is hiding from us. Below you will find some code that I wrote do to demonstrate the issue. I had to do the system_abort as I cannot debug my custom board without a coredump.
`// Define a volatile pointer to the register
volatile uint32_t *reg_ptr = (volatile uint32_t *)0x6008002C;

// Read the register value
uint32_t reg_value = 0;

value2_4 = txsts_bm->fifo_available;
value3_4 = txsts_bm->req_queue_available;
value4_4 = (uint32_t)txsts_bm;
reg_value = *reg_ptr;
// Print the register value in hexadecimal format
value1_4 = reg_value;
value5_4 = ((volatile dwc2_hptxsts_t*)reg_ptr)->fifo_available;
value6_4 = ((volatile dwc2_hptxsts_t*)reg_ptr)->req_queue_available;

// Define a volatile pointer to the register
//volatile uint8_t *reg_ptr1 = (volatile uint32_t *)0x6008002D;
uint16_t reg_value1 = 0;
reg_value1 = *(uint16_t *)reg_ptr;
value7_4 = reg_value1;

esp_system_abort("handle_txfifo_empty core dump trigger");
`

and here comes the code, that the compiler generates:
flash.text:42010629                 andi    a9, a9, 0xFF
.flash.text:4201062C                 memw
.flash.text:4201062F                 l8ui    a8, a10, 1
.flash.text:42010632                 andi    a8, a8, 0xFF
.flash.text:42010635                 slli    a8, a8, 8
.flash.text:42010638                 or      a8, a8, a9
.flash.text:4201063B                 l32r    a9, off_42000D64 ; value2_4 = 0
.flash.text:4201063E                 memw
.flash.text:42010641                 s32i.n  a8, a9, 0
.flash.text:42010643                 memw
.flash.text:42010646                 l8ui    a8, a10, 2      ; <= byte access to hardware register, results in reading 0
.flash.text:42010649                 andi    a8, a8, 0x7F
.flash.text:4201064C                 l32r    a9, off_42000D68 ; value3_4 = 0
.flash.text:4201064F                 memw
.flash.text:42010652                 s32i.n  a8, a9, 0
.flash.text:42010654                 l32r    a8, off_42000D6C ; value4_4 = 1611137068 (0x6008002C)
.flash.text:42010657                 memw
.flash.text:4201065A                 s32i.n  a10, a8, 0
.flash.text:4201065C                 l32r    a8, dword_42000D00
.flash.text:4201065F                 memw
.flash.text:42010662                 l32i.n  a10, a8, 0
.flash.text:42010664                 l32r    a9, off_42000D70 ; value1_4 = 262176 (0x40020)
.flash.text:42010667                 memw
.flash.text:4201066A                 s32i.n  a10, a9, 0
.flash.text:4201066C                 memw
.flash.text:4201066F                 l32i.n  a9, a8, 0
.flash.text:42010671                 andi    a9, a9, -1
.flash.text:42010674                 l32r    a10, off_42000D74 ; value5_4 = 32
.flash.text:42010677                 memw
.flash.text:4201067A                 s32i.n  a9, a10, 0
.flash.text:4201067C                 memw
.flash.text:4201067F                 l32i.n  a9, a8, 0
.flash.text:42010681                 extui   a9, a9, 0x10, 7
.flash.text:42010684                 l32r    a10, off_42000D78 ; value6_4 = 4
.flash.text:42010687                 memw
.flash.text:4201068A                 s32i.n  a9, a10, 0
.flash.text:4201068C                 l16ui   a9, a8, 0       ; <= word access to hardware register, results in reading 0
.flash.text:4201068F                 l32r    a8, off_42000D7C ; value7_4 = 0
.flash.text:42010692                 memw
.flash.text:42010695                 s32i.n  a9, a8, 0
.flash.text:42010697                 l32r    a10, off_42000D80 ; "handle_txfifo_empty core dump trigger"
.flash.text:4201069A                 l32r    a8, off_42000D0C
.flash.text:4201069D                 callx8  a8              ; esp_system_abort

===
The problem is, that the compiler generates code, that performs byte access to a hardware register that has to be accessed as a 32-bit read. When the hardware register is read as byte or as word, the value will always be 0. Please note, that eps32-s3 and esp32-p4 have a different hardware architecture (xtensa vs risc-v). So the solution should be to make a 32-bit read to the hardware register and mask the according bits of interest afterwards.

HiFiPhile, 2025-03-27:
Nice found, it looks like something similar to #2940, in that case it's a armcloang's bug. I'm not familiar with the ABI of xtensa core.

torsts, 2025-03-27:
@HiFiPhile  . yes, pretty similiar.. and I agree to the MACRO strategy as I have seen something like this before. Unfortunately I am a reverse engineer and not a coder.. otherwise I would go and shoot for a fix myself.

HiFiPhile, 2025-03-29:
ESP-IDF also met this issue: https://github.com/espressif/esp-idf/blob/a45d713b03fd96d8805d1cc116f02a4415b360c7/components/hal/platform_port/include/hal/misc.h#L34

To be able to use this macro GCC's [statement expression](https://gcc.gnu.org/onlinedocs/gcc/Statement-Exprs.html) and `typeof` extensions are needed, armclang and IAR EWARM(not affected) support it.

PR #3070 (2025-04-15): host/dwc2: fix bitfields access width
**Describe the PR**
Fix gcc-riscv issue about volatile bitfields are incorrectly accessed using an access width not appropriate to the type of its container.

Tested on `espressif_s3_devkitm` with `device_info` and  `cdc_msc_freertos` examples.

For the host I've tested an usb key and an FTDI usb-uart.
```
Device 1: ID 0403:6001 SN FT1ZK9MQ
Device Descriptor:
  bLength             18
  bDescriptorType     1
  bcdUSB              0200
  bDeviceClass        0
  bDeviceSubClass     0
  bDeviceProtocol     0
  bMaxPacketSize0     8
  idVendor            0x0403 [... cut]
closed 2025-04-15 by hathach