hathach/tinyusb #3054: STM32H747 + USB3320C

eng-software, 2025-03-28:
### Operating System

Others

### Board

STM32H747I-DISCO

### Firmware

examples/host/cdc_msc_hid

### What happened ?

I am trying to use STM32H747I-DISCO as Host to plug USB HUB
In this hub I will conect a CDC, HID and MSC.
I solved many problems, but this last one I need help.

The USB3320 CPEN pin controls VBUS and I cant turn this pin on.

### How to reproduce ?

Just run the STM32Cube project
https://github.com/eng-software/TestTinyUSB.git

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

TinyUSB Host CDC MSC HID Example
USBH init on controller 1, speed = High
sizeof(usbh_device_t) = 86
sizeof(hcd_event_t) = 12
sizeof(_ctrl_xfer) = 20
sizeof(tuh_xfer_t) = 24
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 20
CDC init
sizeof(cdch_interface_t) = 196
MSC init
sizeof(msch_interface_t) = 52
sizeof(msch_epbuf_t) = 48
HID init
sizeof(hidh_interface_t) = 14
HUB init
guid, gsnpsid, ghwcfg1, ghwcfg2, ghwcfg3, ghwcfg4
0x00002300, 0x4F54330A, 0x00000000, 0x229FE190, 0x03B8D2E8, 0xE3F00030
Fullspeed PHY init
DMA = 0

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

eng-software, 2025-03-31:
Now its working.
- I fixed my hardware startup
- Removed the option 

Sugestion:
USB disconnection is not working verywell, so I suggest  the change above:

```
static void handle_hprt_irq(uint8_t rhport, bool in_isr) {
  dwc2_regs_t* dwc2 = DWC2_REG(rhport);
  uint32_t hprt = dwc2->hprt & ~HPRT_W1_MASK;
  const dwc2_hprt_t hprt_bm = dwc2->hprt_bm;

  if (dwc2->hprt & HPRT_CONN_DETECT) {
    // Port Connect Detect
    hprt |= HPRT_CONN_DETECT;

    if (hprt_bm.conn_status) {
      hcd_event_device_attach(rhport, in_isr);
    } else {
      hcd_event_device_remove(rhport, in_isr);
    }
  }

  if (dwc2->hprt & HPRT_ENABLE_CHANGE) {
    // Port enable change
    hprt |= HPRT_ENABLE_CHANGE;

    if (hprt_bm.enable) {
      // Port enable
      const tusb_speed_t speed = hprt_speed_get(dwc2);
      port0_enable(dwc2, speed);
    } else {
      // TU_ASSERT(false, );
    	hcd_event_device_remove(rhport, in_isr);
    }
  }

  dwc2->hprt = hprt; // clear interrupt
}
```

Added hcd_event_device_remove(rhport, in_isr); above  // TU_ASSERT(false, );

HiFiPhile, 2025-04-06:
You can try https://github.com/hathach/tinyusb/pull/2960/commits/bfa971ac08d9decd1e98f888c135d140b20489cd from #2960