hathach/tinyusb #3071: [dwc2] bInterval time for interrupt endpoint not respected

maximevince, 2025-04-09:
### Operating System

Linux

### Board

STM32F7

### Firmware

tinyUSB master, USB host on STM32F7 w/ DWC USB IP Core

### What happened ?

When connecting a HS device with an interrupt endpoint with `bInterval = 1` (1 microframe, 125us), it is being polled only at 2 microframe interval (250us), in case of a NAK.

### How to reproduce ?

Connected a device with bInterval = 1 on it's interrupt endpoint

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Debug log not relevant, I think.

The issue seems to be in `handle_channel_in_slave` -> `channel_xfer_in_retry`.

### Screenshots

[image]

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

maximevince, 2025-04-09:
What I am seeing is this alternating interval of:
- the channel IRQ finding a NAK (+HALTED), then deferring issuing a new IN token to the SOF handler. (every 250us)
- then 1 microframe later: the SOF IRQ re-scheduling the IN token, (so also every 250us, but with 125us offset to the channel IRQ)

[image]

It seems to me that the channel IRQ handler should immediately re-issue a new IN token (submit URB) after receiving the NAK, such that on the next microframe we can get the ACK / NACK response.

maximevince, 2025-04-09:
I've found a potential solution by calling `edpt_xfer_kickoff()` from the `channel_xfer_in_retry()` handler, for `hcint & HCINT_HALTED`.
This is essentially the same thing as the SOF handler would do, but then right after receiving the NACK, instead of wasting a microframe waiting for the next SOF to come along.

Any comments on what The Right Way to do this, is?

HiFiPhile, 2025-04-11:
I need to do some tests to understand better periodic transfer... With 1st look it seems not only `bInterval = 1` but all bInterval have an extra SOF period.

closed 2025-04-24 by hathach