hathach/tinyusb #3073: Repeated hub connect events can leak devices

tannewt, 2025-04-10:
### Operating System

Linux

### Board

Fruit Jam

### Firmware

CircuitPython using TinyUSB commit: 60e6d53d1

### What happened ?

An assertion occurred when TinyUSB ran out of device numbers.

### How to reproduce ?

Plug in the LS mouse from the Adafruit shop: https://www.adafruit.com/product/2025 into the Fruit Jam.

For some reason, the onboard hub detects brief USB device disconnects. It emits the port connection change each time it does. By the time the status is read the connection state is 1 again. This triggers TinyUSB to reset the port and start enumeration:
https://github.com/hathach/tinyusb/blob/865e3488f99209c57b73953428dccf2bc666f0be/src/host/hub.c#L432-L446

However, because connection is 1 again, it never detaches the old device on the port. This leaks device numbers and generally device state. The device itself won't respond because it's been reset and given a new, different device number.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

This is just selective logging of the status change bits and status report for the port.

```
Serial console setup
[1:] USBH Device Attach
  Hub Status Change = 0x08
  HUB connection change addr = 5 port = 3
  BITS change 0001 status 0301
  HUB ccccc addr = 5 port = 3
[1:] USBH Device Attach
lun 1 is not writable
lun 1 is not writable
lun 1 is not writable
lun 1 is not writable
  Hub Status Change = 0x08
  HUB connection change addr = 5 port = 3
  BITS change 0001 status 0301
  HUB ccccc addr = 5 port = 3
[1:] USBH Device Attach
  Hub Status Change = 0x08
  HUB connection change addr = 5 port = 3
  BITS change 0001 status 0301
  HUB ccccc addr = 5 port = 3
[1:] USBH Device Attach
XR_*=1
  Hub Status Change = 0x08
  HUB connection change addr = 5 port = 3
  BITS change 0001 status 0301
  HUB ccccc addr = 5 port = 3
[1:] USBH Device Attach
XR_*=1

```

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-04-11:
seems like an deboucing issue, I will look at this, and order the mouse for testing if needed

HiFiPhile, 2025-04-11:
Me too I was looking at enumerate handling without hub. Currently there are  2 issues related to blocked debouncing wait:
- `_usbh_q` filled up
- xfer complete event of previous enumeration attempt arrive at next enumeration

I'm testing by break down the delay into smaller chunk, so far it's more reliable:
https://github.com/HiFiPhile/tinyusb/commit/22ea6ee67a4f545409b9290ef57c51e3d02d797a

hathach, 2025-05-05:
still happend with sigma mouse + fruit jam, doing more tests to fix.

closed 2025-05-06 by hathach

hathach, 2025-05-06:
@tannewt @ladyada  I found out that our store sigmachip mouse seems to have an internal timeout. If it does not received HID related request not sure which one (probably get hid report or set protocol/idle), it will reset it self, the reset cause SE0 but seems like not long enough for hub to regconize as disconnect event and then try to enumerate itself.