hathach/tinyusb #3094: Support for CH32 USBFS host mode

joelpmichael, 2025-04-23:
### Related area

New port support

### Hardware specification

CH32 variant with USBFS, eg CH32V203C8T6 as used on the currently supported board ch32v203c_r0_1v0

### Is your feature request related to a problem?

CH32 supports USB Host mode in the USBFS peripheral, but TinyUSB does not support it (yet)

### Describe the solution you'd like

I have started writing support for the CH32 USBFS host mode, and it's now at the point where the USB host will enumerate some devices using the host/device_info example. It seems to enumerate another TinyUSB device using device/cdc_msc, displaying device strings as expected.

There are still timing-related issues in the driver, which is causing enumeration to fail on many devices. I also suspect I have issues with data0/data1 toggling as some devices are reporting STALL during enumeration. The code is available at https://github.com/joelpmichael/tinyusb/tree/ch32-usbfs-host but is definitely not ready for a PR yet. Once I can get it reliably enumerating a bunch of different devices I have (eg keyboard, mouse, mass storage, CDC-ACM, UARTs) I will send the PR through. If anyone wishes to check out the code and join the development, I will happily accept patches because I'm rapidly approaching the limit of my USB knowledge.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

braxlan, 2025-04-23:
Have you seen [PR 2793](https://github.com/hathach/tinyusb/pull/2793)? Chances are high because it is neither an issue, a discussion or part of the documentation.

joelpmichael, 2025-04-23:
I have not seen this one, thank you! I guess keep this bug open until it's merged, so it's a bit more visible that this support exists in the wild?

braxlan, 2025-04-24:
That sounds like a good idea! I would love to see more people getting involved with the CH32 family because of its unique feature of having two USB ports which makes it ideal for dual-role projects.

hathach, 2025-07-05:
implemented byy https://github.com/hathach/tinyusb/pull/2793

closed 2025-07-05 by hathach