hathach/tinyusb #3098: Handle suspend events due to EMI

marchingband, 2025-04-24:
### Operating System

Windows 11

### Board

esp32-s3

### Firmware

```
void usb_start(void){
    ESP_LOGI(TAG, "USB Composite initialization");
    const tinyusb_config_t tusb_cfg = {
        .device_descriptor = NULL,
        .string_descriptor = NULL,
        .string_descriptor_count = 0,
        .external_phy = false,
        .configuration_descriptor = NULL,
    };
    ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));

    tinyusb_config_cdcacm_t acm_cfg = {
        .usb_dev = TINYUSB_USBDEV_0,
        .cdc_port = TINYUSB_CDC_ACM_0,
        .rx_unread_buf_sz = 64,
        .callback_rx = &tinyusb_cdc_rx_callback,
        .callback_rx_wanted_char = NULL,
        .callback_line_state_changed = tinyusb_cdc_line_state_changed_callback,
        .callback_line_coding_changed = tinyusb_cdc_line_coding_changed_callback
    };

    ESP_ERROR_CHECK(tusb_cdc_acm_init(&acm_cfg));

    ESP_LOGI(TAG, "USB Composite initialization DONE"); 
}
```

### What happened ?

I am using this CDC device driver in a context where there is sometimes a lot of EMI.
When this EMI causes a problem, the host will suspend the CDC device, and the driver provides callbacks for this, but I have been unable to find a way to restart, reconnect and have the host enumerate again.
I believe there was some work on `bool tusb_teardown(void);`, is there a recommended way to handle this situation?
Apologies if this is not a "bug", although arguably the need to restart the stack from scratch is either a feature request or a bug?
thank you so much!

### How to reproduce ?

I have a cheap inverter, which produces a lot of EMI, and a USB cable without chokes on it. In this scenario the EMI will periodically cause the suspend event.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

-

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

FurTrader, 2025-04-24:
To clarify, without the chokes on the USB cable, turning on the inverter immediately interrupts the USB communication.

With 2 chokes on the cable, the connection is stable for a random amount of time. Sometimes all day, sometimes a few minutes.

With no EMI generating loads switched on, the connection is stable indefinitely.

hathach, 2025-05-21:
try to use tud_disconnect() delay 10ms and tud_connect(). Closed since it is not a bug

closed 2025-05-21 by hathach