hathach/tinyusb #3108: Support ESP32-P4 USB Host

robin25, 2025-05-01:
### Related area

I would like to have tinyUSB support USB Host on ESP32-P4

### Hardware specification

ESP32-P4

### Is your feature request related to a problem?

ESP-IDF provide is own usb host stack but it only support HS with no STT support for FS or LS device connected on HUB.

It would be benefical if tinyusb support esp32p4 as tinyusb support LS FS and HS device on hub. 

### Describe the solution you'd like

Add host support for the P4, and ideally an example.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2025-05-05:
already supported

closed 2025-05-05 by hathach

RoxerRN, 2025-05-07:
Can you please elaborate a bit more please ? 

For most of people USB is a bit complicated, and most people on esp32.com argue that tinyusb do not support host mode on the P4.  (all idf exemples with tinyusb are for usb device only, the host examples are made with the espressif host stack ( which is limited, and do not support LS ans FS device on a FS HUB)). 

There seems to be a discrepancy between the TinyUSB documentation and the advice provided on the ESP32 forum. I saw on the main page of the tinyusb project that tinyusb support Device, Host mode and HS speed but nowhere I can find an example of how it's possible to set the P4 (or S3 btw) as an USB host. 

Also the espressif documentation detail a bit how to work with stock TinyUSB but it's far from intuitve to me : https://docs.espressif.com/projects/esp-iot-solution/en/latest/usb/usb_overview/tinyusb_development_guide.html

And esp provide : https://components.espressif.com/components/espressif/tinyusb/versions/0.18.0~2 which is 5 commit behind TinyUSB repo. So I made the assumption that the idf component support all the functionality provider by stock TinyUBS, right ? 

So a lot of questions remains : 

Is Tinyusb provide dwc2 driver for the P4 as host ?
Can we work with the 1.1 peripheral and the HS peripheral ?

Can you please provide some insight please ? 

Best regards,

roma-jam, 2025-05-07:
Hi @RoxerRN , 

> all idf exemples with tinyusb are for usb device only

In esp-idf we use the tinyUSB for device examples only, as we don't have our own device stack. But that doesn't mean, that tinyUSB doesn't support host stack. 

> host examples are made with the espressif host stack ( which is limited, and do not support LS ans FS device on a FS HUB)

Currently, the limitation for Full-speed USB peripheral was solved (https://github.com/espressif/esp-idf/issues/15683) and now there is a possibility to attach the Low- and Full-speed devices via external hubs to Full-speed USB peripheral (this is applied for S2/S3 and P4 Full-speed peripheral)

USB High-speed peripheral on P4 requires the support of split transactions. Split transactions are not supported in Scatter-Gather DMA mode (which is used in the espressif host stack), but they are supported in the two other modes: Slave and Buffer DMA. 

So, with tinyUSB host stack you are able to attach the Low- or Full-speed devices to USB High-speed peripheral on P4.

You can use the example from tinyusb: https://github.com/hathach/tinyusb/tree/master/examples/host/device_info, but you need config the right port (0), config the speed (High) and enable the UTMI phy interface in [family.c](https://github.com/hathach/tinyusb/blob/master/hw/bsp/espressif/boards/family.c#L172): 

```
      // .target = USB_PHY_TARGET_INT,
    .target = USB_PHY_TARGET_UTMI,
```

RoxerRN, 2025-05-08:
Hi @roma-jam Thank you very much for your answers, 

> So, with tinyUSB host stack you are able to attach the Low- or Full-speed devices to USB High-speed peripheral on P4.

I definitely want to give it a try, my board is built this way. I managed to compile TinyUSB with the P4 but I cannot manage to compile family.c because #include "bsp/board_api.h" #include "board.h" are not found. I use idf-esp to compile so I think i cannot use family.cmake and so one. 

Do you know the proper way to compile TinyUSB with the right bsp ? Is it the right path to compile with the BSP provided by TinyUSB or should I implement it myself ?

roma-jam, 2025-05-14:
> Do you know the proper way to compile TinyUSB with the right bsp ? 

There is such topin in the esp32 forum: https://esp32.com/viewtopic.php?p=147573#p147573

RoxerRN, 2025-05-15:
Hi Roma, 

It is my topic on the esp32 forum you mentioned, as explained in the post I actually managed to compile tinyusb with esp-idf. It works like a charm and in the end it's not so difficult when you understand how to compile it with the module provided by espressif.

I don't know if the post is sufficient to help others to get it compile on the p4, or if it's can be useful if I provide somewhere a full working example ? ( if you know a good place to store such example I can provide a sample compiling project) 

Thanks again,