hathach/tinyusb #3112: Unable to enumerate on audio examples with STM32WBxx

P3Vu, 2025-05-05:
### Operating System

Linux

### Board

STM32WB55

### Firmware

Master branch with following examples:

examples/device/uac2_headset
examples/device/audio_test
examples/device/uac2_speaker_fb

### What happened ?

Recently I tried to run audio examples on my dev board STM32WB55 and figured out that host is unable to enumerate the device properly. I checked previous versions of the library and figure out that on tag 0.16.0 for audio_test example problem don't exist so I went through changes introduced since then and find out that issue started at commit 1cf8e34ae5fc6466d07ac438691391c7a5a809ff 

**dmesg logs**
[10201.413752] usb 1-8: new full-speed USB device number 35 using xhci_hcd
[10201.527781] usb 1-8: device descriptor read/64, error -71
[10201.746703] usb 1-8: device descriptor read/64, error -71
[10201.962739] usb 1-8: new full-speed USB device number 36 using xhci_hcd
[10202.076687] usb 1-8: device descriptor read/64, error -71
[10202.300679] usb 1-8: device descriptor read/64, error -71
[10202.408846] usb usb1-port8: attempt power cycle
[10202.796719] usb 1-8: new full-speed USB device number 37 using xhci_hcd
[10202.796916] usb 1-8: Device not responding to setup address.
[10203.001756] usb 1-8: Device not responding to setup address.
[10203.208725] usb 1-8: device not accepting address 37, error -71
[10203.208824] usb 1-8: WARN: invalid context state for evaluate context command.
[10203.322719] usb 1-8: new full-speed USB device number 38 using xhci_hcd
[10203.322916] usb 1-8: Device not responding to setup address.
[10203.528740] usb 1-8: Device not responding to setup address.
[10203.736631] usb 1-8: device not accepting address 38, error -71
[10203.736714] usb 1-8: WARN: invalid context state for evaluate context command.
[10203.736786] usb usb1-port8: unable to enumerate USB device
`

Checking out on one commit before this allows to enumerate the mic for audio_test but still there is some problem on the rest of the examples. 

### How to reproduce ?

Build and flash mentioned examples on STM32WBxx board with on master branch.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

n/a

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

P3Vu, 2025-05-10:
I connected a debugger and problem appears in `pcd_get_endpoint` and `pcd_get_endpoint`:

[image]

According to stm32wb55 manual USBFS should point to 0x40006800:

[image]

I hardcoded FSDEV_REG to the previous value and it works both on that commit and current master branch. All mentioned examples are enumerating correctly and appearing as devices in the host system.

`#define FSDEV_REG ((fsdev_regs_t*) 0x40006800)`

HiFiPhile, 2025-05-10:
Thank you for looking at it, FSDEV_BASE is define by ST's header file, maybe is a ST's issue.

https://github.com/hathach/tinyusb/blob/542e5b4550a01d034b78308d77c408ed89427513/src/portable/st/stm32_fsdev/fsdev_stm32.h#L219

P3Vu, 2025-05-10:
It appears that tinyUSB uses v.10.0 of **cmsis-device-wb** (Nov 29,2021) and relevant address changed to proper value on v1.12.0 (March 10, 2023), we could bump to latest available v.12.2 (Jul 2, 2024) in `get_deps`.

I've done it locally and observe no issues.

P3Vu, 2025-05-15:
Fixed here https://github.com/hathach/tinyusb/pull/3121 . Closing