hathach/tinyusb #3116: VENDOR interface enumeration error switching from linux host to virtualbox windows guest

wndmr, 2025-05-07:
### Operating System

Linux

### Board

rp2040

### Firmware

webusb_serial example in picosdk_v2.1.0 compiled with LOG=2
Developement setup: ubuntu24.04 host computer, Windows10 guest in virtualbox.

### What happened ?

Enumeration fails when usb device is assigned to virtualbox guest.

### How to reproduce ?

Once the device get connected to the host computer it enumerates as normal device accessible from host applications. Now assigning the device to virtualbox guest system the enumeration fails in Windows 10.

The issue arises at vendor_device.c 
```
    bool tud_vendor_n_mounted(uint8_t itf) {
      TU_VERIFY(itf < CFG_TUD_VENDOR);
      vendord_interface_t* p_itf = &_vendord_itf[itf];
      return p_itf->rx.stream.ep_addr || p_itf->tx.stream.ep_addr;
    }
```
as the ep_addr 's are already set (leftover from enumeration on linux)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Upload of txt is not working therefore I past the log here.

                                                                #######################################
                                                                Enumeration in LINUX host
                                                                #######################################

USBD Suspend : Remote Wakeup = 0
USBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 84
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 20
sizeof(tu_edpt_stream_t) = 36
CDC init
VENDOR init
Chip Version B2
sizeof(hw_endpoint_t) = 32
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 12 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 0F 00 00 05 00
  Get Descriptor BOS
  Queue EP 80 with 5 bytes ...
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 0F 00 00 39 00
  Get Descriptor BOS
  Queue EP 80 with 57 bytes ...
USBD Xfer Complete on EP 80 with 57 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 02 00 00 09 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes ...
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 62 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes ...
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 30 bytes ...
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 FF 00
  Get Descriptor String[1]
  Queue EP 80 with 16 bytes ...
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 FF 00
  Get Descriptor String[3]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00
  Set Configuration
  Open EP 81 with Size = 8
  Allocated 64 bytes at offset 0x180 (0x50100180)
  Open EP 02 with Size = 64
  Allocated 128 bytes at offset 0x1c0 (0x501001C0)
  Open EP 82 with Size = 64
  Allocated 128 bytes at offset 0x240 (0x50100240)
  Queue EP 02 with 64 bytes ...
  CDC opened
  Bind EP 81 to driver id 0
  Bind EP 02 to driver id 0
  Bind EP 82 to driver id 0
  Open EP 03 with Size = 64
  Allocated 128 bytes at offset 0x2c0 (0x501002C0)
  Queue EP 03 with 64 bytes ...
  Open EP 83 with Size = 64
  Allocated 128 bytes at offset 0x340 (0x50100340)
  VENDOR opened
  Bind EP 03 to driver id 1
  Bind EP 83 to driver id 1
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 04 03 09 04 FF 00

                                                                ######################################
                                                                Following Enumeration in Windows guest
                                                                after assigning usb device to virtual
                                                                box guest
                                                                ######################################
USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 09 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes ...
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 62 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 12 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 0F 00 00 05 00
  Get Descriptor BOS
  Queue EP 80 with 5 bytes ...
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 0F 00 00 39 00
  Get Descriptor BOS
  Queue EP 80 with 57 bytes ...
USBD Xfer Complete on EP 80 with 57 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 62 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 FF 00
  Get Descriptor String[3]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00
  Set Configuration
  Open EP 81 with Size = 8
  Allocated 64 bytes at offset 0x180 (0x50100180)
  Open EP 02 with Size = 64
  Allocated 128 bytes at offset 0x1c0 (0x501001C0)
  Open EP 82 with Size = 64
  Allocated 128 bytes at offset 0x240 (0x50100240)
  Queue EP 02 with 64 bytes ...
  CDC opened
  Bind EP 81 to driver id 0
  Bind EP 02 to driver id 0
  Bind EP 82 to driver id 0
process_set_config 1049: ASSERT FAILED                      !!!!!! unexpected behaviour
process_control_request 789: ASSERT FAILED                  !!!!!! unexpected behaviour
  Stall EP0
USBD Bus Reset : Full Speed

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-05-07:
Please update to master manually, picosdk is too old. 53989a9

wndmr, 2025-05-08:
... problem is gone with current master branch. Thank you for help.

closed 2025-05-08 by wndmr