hathach/tinyusb #3142: uac2_headset no sound card

karu2003, 2025-06-09:
### Operating System

Linux

### Commit SHA

86ad6e56c

### Board

Raspberry Pi Pico 

### Firmware

uac2_headset

### What happened ?

build this project. sound card did not appear. I work under Ubuntu. 
https://github.com/hathach/tinyusb/tree/master/examples/device/uac2_headset

alsamixer shows nothing. :( 

[  184.366820] usb 1-4: Product: TinyUSB headset
[  184.366822] usb 1-4: Manufacturer: TinyUSB
[  184.366824] usb 1-4: SerialNumber: E6605838835DC22F
[  184.453661] mc: Linux media interface: v0.10
[  199.844950] usb 1-4: parse_audio_format_rates_v2v3(): unable to retrieve number of sample rates (clock 4)
[  204.964432] usb 1-4: parse_audio_format_rates_v2v3(): unable to retrieve number of sample rates (clock 4)
[  215.204915] usb 1-4: 2:0: failed to get current value for ch 1 (-110)
[  225.445481] usb 1-4: 2:0: failed to get current value for ch 2 (-110)
[  235.685296] usb 1-4: 2:0: failed to get current value for ch 0 (-110)
[  245.924841] usb 1-4: 2:0: cannot get min/max values for control 2 (id 2)
[  251.044650] usb 1-4: 2:0: cannot get min/max values for control 2 (id 2)
[  251.044887] usbcore: registered new interface driver snd-usb-audio
[  257.188813] usb 1-4: 2:0: cannot get min/max values for control 2 (id 2)
[  262.308582] usb 1-4: 2:0: cannot get min/max values for control 2 (id 2)
[  267.428427] usb 1-4: 2:0: cannot get min/max values for control 2 (id 2)

### How to reproduce ?

build and copy 

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

not Debug. :( 

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-06-09:
Dump of #3110

karu2003, 2025-06-09:
Does this example work at 133 or 250mhz?

ashafq, 2025-10-02:
I have the same issue with a Pico2 as well. Can anyone suggest a fix?

HiFiPhile, 2025-10-02:
> [@HiFiPhile](https://github.com/HiFiPhile) I am currently on the 2.1.1 sdk path, and confirmed that I am on 0.18.0. It just panics upon boot. So, there's something definitely wrong:

[image]