hathach/tinyusb #3146: Raw HID TUSB:DCD Complete but not empty error, SendReport() report 1 wait failed blocks outgoing data

jrowberg, 2025-06-13:
### Operating System

Windows 11

### Commit SHA

Version bundled in framework-arduinoespressif32 version 3.20017.241212+sha.dcc1105b, "a0e5626bc" mentioned in docs

### Board

Custom ESP32-S2 design, wifi inactive

### Firmware

Custom firmware in PlatformIO env, sending 2x 64-byte raw HID packets every second. I haven't created a new minimal test case project yet, but I will try do so shortly.

### What happened ?

Periodic HID output reports (from device to host PC) are sent in sets of 2, one right after the other, once per second. This works perfectly for between 30 seconds and 12 hours--maybe more sometimes, I haven't always been capturing log data. But then at some point, a `Complete but not empty: 0/64` error is generated in `TUSB:DCD` code, and immediately afterwards, the next `SendReport()` call results in `report 1 wait failed`, and no more data can be sent out. Incoming data from the host PC to the device still flows normally. This condition is not fixed until I reset the MCU.

**ASIDE FROM THE DATA IN THIS BUG REPORT**, what I would most like to do is learn how to find and debug the code that generates the initial `Complete but not empty: 0/64` error. I haven't been able to find it yet; it seems to be a message present only in the precompiled `libarduino_tinyusb.a` file in `.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\lib\`. I will gladly flesh out the bug report with more info if I can dig and test more. At the moment, I'm flying blind and only able to observe one (blocking) symptom of the problem.

### How to reproduce ?

I don't have to do anything special in the custom firmware, just let it run until the problem appears. As described elsewhere, it is simply sending two 64-byte raw HID packets to the host every second; nothing really special. As soon as I get minimal test project doing the same thing, I will post more detail.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Here is an example log capture, though this doesn't have as much as I would like to see. I have added the `xSTres=%d` output temporarily in USBHID.cpp to confirm the correct return code from the first `xSemaphoreTake` call in `SendReport()`, as I suspect some kind of race condition at the moment.

```
[2025-06-12 12:17:34.259] [1598029][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:34.259] [1598037][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:35.269] [1599040][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:35.269] [1599049][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:36.281] [1600051][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:36.281] E (1533379) TUSB:DCD: Complete but not empty: 0/64
[2025-06-12 12:17:36.281] [1600060][I][USBHID.cpp:353] SendReport(): xSTres=0
[2025-06-12 12:17:36.390] [1600170][E][USBHID.cpp:360] SendReport(): report 1 wait failed
...[no more outgoing SendReport() events]
```

The raw HID report uses Report ID 1. I have tried shrinking the raw HID report from 64 down to 32 bytes on a whim, but the problem still occurs.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2025-06-13:
Please open the issue in corresponding repo, `Complete but not empty: 0/6` or `TUSB:DCD` is not covered here.

jrowberg, 2025-06-13:
> Please open the issue in corresponding repo, `Complete but not empty: 0/6` or `TUSB:DCD` is not covered here.

I'm more than happy to do this, but can you point me towards the right repo? I feel dumb asking, but this was my best guess based on my understanding of the TinyUSB implementation for the ESP32/Arduino framework, and the log tag found here:

https://github.com/hathach/tinyusb/blob/e95973d3462cdacd165a4057632d46248ded7b8a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L76

HiFiPhile, 2025-06-13:
It looks like from https://github.com/espressif/esp32-arduino-lib-builder but `dcd_esp32sx` is obsolete and removed. 

You can find it in older forks https://github.com/esp-arduino-libs/esp32-arduino-lib-builder/blob/c79c44c198629d3a4b28395e8aa73750403964bb/components/arduino_tinyusb/src/dcd_esp32sx.c#L805