hathach/tinyusb #3154: ESP32-S3 bulk transfer issues

qnex0, 2025-06-25:
### Operating System

Linux

### Commit SHA

 espressif/tinyusb (0.18.0~2)

### Board

ESP32-S3 based board

### Firmware

[tinyusb_firmware.zip](https://github.com/user-attachments/files/20912887/tinyusb_firmware.zip)

[libusb_test.zip](https://github.com/user-attachments/files/20912889/libusb_test.zip)

### What happened ?

I'm trying to implement a custom vendor class driver. I noticed that when using usbd_edpt_xfer to receive data from the OUT EP, the ZLP is misinterpreted and causes the transfer to complete with an xfer size equal to the request size regardless of the actual size. I tried some older versions of tinyusb and noticed this behavior started occurring after version "0.15.0~10." Additionally, when I queue for 32768 bytes in my example, when more than 8191 bytes are sent to the device, the transfer is also completed with the incorrect size.

### How to reproduce ?

I use the libusb_test program attached to the firmware section of this issue to send a variety of different data sizes to the bulk endpoint and view the results using the firmware with debugging enabled.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A

### Screenshots

[image]

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

dpgeorge, 2025-11-12:
Just for reference, it looks like this commit db7670a3bcd502193d138041c20d401bc3f59001 was responsible for this issue (it seems that ZLP was correctly handled prior to that commit).

hathach, 2025-11-17:
@dpgeorge it seems to be fixed by #3293, let me know if that works for you.

dpgeorge, 2025-11-17:
@hathach yes, that PR fixes the issue.  We have needed to update our TinyUSB submodule to include that patch which is not yet in a release (see https://github.com/micropython/micropython/pull/18406) and also fork the Espressif fork of TinyUSB to patch that as well (see https://github.com/micropython/micropython/pull/18407).

hathach, 2025-11-20:
0.20.0 is release, please give it a try to see if there is any other new issues.