hathach/tinyusb #3192: USB Host: Add support for Adafruit USB/Serial Converter (PID 18)

makermelissa, 2025-07-31:
### Related area

USB Host detection

### Hardware specification

RP2040

### Is your feature request related to a problem?

I am trying to enumerate the device using the arduino Dual Role device_info example. Here is the descriptor info that I pulled off using a computer:

```
Information for device FT232R USB UART (VID=0x0403 PID=0x6001):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0001
Current configuration value: 0x00
Number of open pipes: 0

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x0403	idVendor
0x6001	idProduct
0x0600	bcdDevice
0x01	iManufacturer   "FTDI"
0x02	iProduct        "FT232R USB UART"
0x03	iSerialNumber   "A979L8FK"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0020	wTotalLength   (32 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0x2D	bMaxPower      (90 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0xFF	bInterfaceSubClass   
0xFF	bInterfaceProtocol   
0x02	iInterface   "FT232R USB UART"

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Microsoft OS Descriptor:
------------------------------
0x00	bLength
0x00	bDescriptorType
Hex dump: 

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "FTDI"
0x02   0x0409  "FT232R USB UART"
0x03   0x0409  "A979L8FK"

------------------------------

Connection path for device: 
Intel(R) ICH9 Family USB Universal Host Controller - 2934
Root Hub
FT232R USB UART (VID=0x0403 PID=0x6001) Port: 1
```

### Describe the solution you'd like

I would like it supported so I can connect tor RS-232 devices using the USB Host.

### I have checked existing issues, dicussion and documentation

- [x] I confirm I have checked existing issues, dicussion and documentation.

makermelissa, 2025-08-06:
I'll give that a try. Setup is just a feather rp2040 with usb host using arduino.

hathach, 2025-08-07:
@makermelissa ok, I will pull out the hw to test and post updat here.

hathach, 2025-09-05:
@makermelissa https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/551 pull the latest from tinyusb, this should fix the issue with ftdi driver. Please give it a try and let me know if that works for you.