hathach/tinyusb #3199: Pure Vendor Specific Descriptor breaks enumeration on Linux/Android on CH32V305

AdrianTunkin, 2025-08-04:
### Operating System

Linux

### Commit SHA

87d9e05efa951d6e1b1c63cbb7f4261be96741e1

### Board

nanoCH32V305

### Firmware

[CH32V305RBT6_pure_vendor_specific.zip](https://github.com/user-attachments/files/21564031/CH32V305RBT6_pure_vendor_specific.zip)

### What happened ?

**Summary** 
When connecting a CH32V305-based USB device using TinyUSB to an Android host, the enumeration process:
      Takes abnormally long (≈8–10 seconds)
      Fails to show Manufacturer and Product Name in system/device UI

However:
    On Windows, the same device enumerates quickly and displays all strings correctly (see Windows_USB_Device_Viewer.txt )

If I remove the vendor-specific interface (TUD_VENDOR_DESCRIPTOR) from the USB descriptor, the enumeration on Android is fast and all strings appear as expected.

**Environment**
    MCU: CH32V305 (but CFG_TUSB_MCU is set to OPT_MCU_CH32V307)
    USB Speed: High Speed
    TinyUSB Class Used: CFG_TUD_VENDOR = 1
    Endpoints: 0x81 (IN) and 0x01 (OUT)
    Host: Android (Linux), various devices tested
    TinyUSB Log Level: 2

**Observations from TinyUSB Logs**
  With the vendor-specific interface defined:
  
  USBD Setup Received 00 09 01 00 00 00 00 00 
    Set Configuration
    Open EP 81 with Size = 64
  tu_edpt_validate 245: ASSERT FAILED

This shows that enumeration reaches the Set Configuration stage, but TinyUSB fails internally trying to open endpoint 0x81.

**Thank team for the project and hard work**

### How to reproduce ?

1. open project in MounRiver Studio (file CH32V305RBT6.wvproj)
2. Compile: Project -> Build Project
3. Flash to nanoCH32V305:
- put board into bootloader mode: press buttons BOOT + RESET
- Open WCHISPStudio, click deprotect
- again put board into bootloader mode: press buttons BOOT + RESET
- choose firmware file: C:\Users\tunki\mounriver-studio-projects\CH32V305RBT6\obj\CH32V305RBT6.hex
- click Download.

4. Now plug cable to USB2-HS port and plug to Android phone via USB-OTG.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[tusb_logs.txt](https://github.com/user-attachments/files/21576976/tusb_logs.txt)

[Windows_USB_Device_Viewer.txt](https://github.com/user-attachments/files/21577170/Windows_USB_Device_Viewer.txt)

### I have checked existing issues, discussion and documentation

- [x] I confirm I have checked existing issues, discussion and documentation.

HiFiPhile, 2025-08-04:
The assert error message is clear enough. See cdc_dual_ports example for how to support both full and high speed.

https://github.com/hathach/tinyusb/blob/87d9e05efa951d6e1b1c63cbb7f4261be96741e1/examples/device/cdc_dual_ports/src/usb_descriptors.c#L180

closed 2025-08-04 by HiFiPhile

AdrianTunkin, 2025-08-05:
Success.
Thank team for the hard work.