hathach/tinyusb #3213: tinyUSBv18 RP2040 on Windows 11 using UAC2 and CDC

mdhosale, 2025-08-13:
### Operating System

Windows 11

### Commit SHA

version 0.18.0 bundled by pico

### Board

RP2040 custom board

### Firmware

full code example is available here: https://github.com/mdhosale/tinyUSBv18bugRP2040
this code is from a larger project but was tested and verified before uploading.

I am using a custom descriptor to get 2-channel out from USB - but from what I can tell it aligns with the format of the  1 and 4 channel descriptor embedded in the API

### What happened ?

UAC2 and CDC composite work with TinyUSB v0.17.0 on windows 11 but not on TinyUSB v0.18.0 \
It does work on MacOS - not tested on Linux
 
- I have tried updating:  

`tusb_init(); `

to 

```
    tusb_rhport_init_t dev_init = {
        .role = TUSB_ROLE_DEVICE,
        .speed = TUSB_SPEED_AUTO
     };
     tusb_init(0, &dev_init); 
```

with no luck - `tusb_init(); ` should be backwards compatible with v0.018

### How to reproduce ?

In the latest SDK the code does not work on Wondows 11.

to fix - I rolled back the TinyUSB Library without rolling back the pico SDK by navigating to:

`../pico-sdk/lib/tinyusb `

and doing the following:

```
git fetch --tags
git checkout 0.17.0
```

make clean and compile and the USB enumerates and runs on Windows 11 (and Mac)

If  I roll back to USB v0.18 by navigating to:

`../pico-sdk/lib/tinyusb `

and doing the following:

```
git fetch --tags
git checkout 0.18.0
```

make clean and compile and the USB enumerates but the audio freezes on Windows 11; while the USB still enumerates and runs on Mac

[... 270 lines cut ...]
bNumEndpoints:                     0x02
bInterfaceClass:                   0x0A  -> This is a CDC Data USB Device Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x04  -> Direction: OUT - EndpointID: 4
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x84  -> Direction: IN - EndpointID: 4
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

### I have checked existing issues, discussion and documentation

- [x] I confirm I have checked existing issues, discussion and documentation.

HiFiPhile, 2025-08-13:
What is `adportaudio_start` ? It doesn't like an error emit by Windows.

The latest master have some RP2040 audio fixes.

mdhosale, 2025-08-13:
Thank you for your reply...

>What is adportaudio_start ? It doesn't like an error emit by Windows.

apologes - I am testing the audio in Max by Cyling74 - this is the error it provided when thw audio froze.

> Your log is hard to read, please use usbtreeview and look error from device manager.
device manager reports the device is working properly

Device USB\VID_CAFE&PID_4011&MI_00\7&332144be&1&0000 was configured.

Driver Name: usbaudio2.inf
Driver Package ID: usbaudio2.inf_amd64_2390f0bffa9b0cab
Class GUID: {4d36e96c-e325-11ce-bfc1-08002be10318}
Driver Date: 07/03/2024
Driver Version: 10.0.26100.1150
Driver Provider: Microsoft
Driver Section: usbaudio2_Device.NT
Driver Rank: 0xFF2003
Matching Device ID: USB\Class_01&SubClass_00&Prot_20
Outranked Drivers: wdma_usb.inf:USB\Class_01:00FFFFFF:USB\Class_01&SubClass_00&Prot_20
Device Updated: false
Parent Device: USB\VID_CAFE&PID_4011\6&c179f6d&0&2
 
**here is the dump from usbtreeview:**

 =========================== USB Port2 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 4-2
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x02 (Port 2)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#5&16684953&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x04 (Port 4)
 -> CompanionPortChain   : 4-4

      ========================== Summary =========================
Vendor ID                : 0xCAFE (Unknown Vendor)
Product ID               : 0x4011
Manufacturer String      : *ERROR* iManufacturer=1 but String Descriptor not found
Product String           : *ERROR* iProduct=2 but String Descriptor not found
Serial                   : *ERROR* iSerial=3 but String Descriptor not found
USB Version              : 2.0 (but 12 Mbit/s FullSpeed only)
Port maximum Speed       : High-Speed (Companion Port 4-4 is doing the SuperSpeed)
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Used Endpoints           : 4
Hint                     : Requesting String Descriptors took 7015 ms

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_CAFE&PID_4011#6&c179f6d&0&2#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-10
Device ID                : USB\VID_CAFE&PID_4011\6&C179F6D&0&2
Hardware IDs             : USB\VID_CAFE&PID_4011&REV_0100 USB\VID_CAFE&PID_4011
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0024 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.26100.3912  Date: 2025-06-13  Company: Microsoft Corporation)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
[... 55 lines cut ...]
bDeviceClass             : 0xEF (Miscellaneous)
bDeviceSubClass          : 0x02
bDeviceProtocol          : 0x01 (IAD - Interface Association Descriptor)
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0xCAFE (Unknown Vendor)
idProduct                : 0x4011
bcdDevice                : 0x0100
iManufacturer            : 0x01 (String Descriptor 1)
 *!*ERROR  String descriptor not found
iProduct                 : 0x02 (String Descriptor 2)
 *!*ERROR  String descriptor not found
iSerialNumber            : 0x03 (String Descriptor 3)
 *!*ERROR  String descriptor not found
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 00 02 EF 02 01 40 FE CA 11 40 00 01 01 02   .......@...@....
                           03 01                                             ..

ConfigDescriptor 1 Error : WAIT_TIMEOUT

      -------------------- String Descriptors -------------------
none

> The latest master have some RP2040 audio fixes.

I thought I was on the latest master, but I will check

HiFiPhile, 2025-08-13:
Lastest master have some fundamental changes to audio class, especially transfers occurs in ISR now.

Please start with one of the example and adapt to your needs, your code is a little too old.

If device manager shows Ok then it looks like some audio specific control request didn't goes well, you can use Wireshark or other capture tools to check the bad request.

mdhosale, 2025-08-13:
Thank you for you quick responses. - I think that's where I am struggling - I noticed the updated doc that added the 

```
void USB0_IRQHandler(void) {
  tusb_int_handler(0, true);
}
```

but I haven't found an example that describes how to implement this on a rp2040. In the example files, as far as I can tell, there isn't any indication of that being implemented. Sorry if I am missing something and thank you again for your help...

HiFiPhile, 2025-08-13:
> Thank you for you quick responses. - I think that's where I am struggling - I noticed the updated doc that added the
>
> ```
> void USB0_IRQHandler(void) {
>   tusb_int_handler(0, true);
> }
> ```
>
> but I [...]

Personally I only use industrially proven MCUs instead of RP2xxx so I can't give you more details. But as far as I can tell pico-sdk handle interrupts differently, you can ignore this part written for general purpose MCU.